jneilliii/OctoPrint-UltimakerFormatPackage #3: Request: UFP images in files list

frezap, 2019-11-03:
I would really apprechiate if the images would be displayed in the files list of octoprint.

To visualize I made a mockup showing how I imagine it to look. This uses your Widescreen plugin to put the files list on the right side. Most other things can be done with some css tweaks and themify.
[image]

jneilliii, 2019-11-03:
Thanks for the mock-up, I see your using my widescreen plugin as well, which removes some of my concerns relative to being cluttered. I'll make it an option, either button with pop-up or inline images.

jneilliii, 2019-11-09:
I haven't been able to add the option to the settings interface, but I do have it where it will read the setting from config.yaml.  SSH to your pi and edit the config.yaml adding the following in the plugins section and then install the url using the link below in plugin manager.

```
plugins:
  UltimakerFormatPackage:
    inline_thumbnail: true
```

https://github.com/jneilliii/OctoPrint-UltimakerFormatPackage/archive/0.1.1.zip

frezap, 2019-11-09:
Thank you so much!!
This is real:
[image]

I couldn't resize the thumbnail with themeify due to a wierd css update bug or whatever. So currently I did it this way:
`			let inline_thumbnail_template = '<div class="row-fluid inline_thumbnail" data-bind="if: ($data.name.indexOf(\'.ufp.gcode\') > -1 && $root.settingsViewModel.settings.plugins.UltimakerFormatPackage.inline_thumbnail() == true)"><img data-bind="attr: {src: $root.inline_thumbnail_url($data)}, visible: ($data.name.indexOf(\'.ufp.gcode\') > -1 && $root.settingsViewModel.settings.plugins.UltimakerFormatPackage.inline_thumbnail() == true), click: function() { if ($root.loginState.isUser()) { $root.open_thumbnail($data) } else { return; } }"` 
`width="100px" style="display: block; margin-left: auto; margin-right: auto"`
`/></div>'`

jneilliii, 2019-11-10:
The selector should be something like `div.inline_thumbnail > img` I would think.

jneilliii, 2019-11-16:
Just released official 0.1.1 version that incorporates a settings page for enabling/disabling the inline thumbnails and some other tweaks.

closed 2019-11-16 by jneilliii

TAOGde, 2020-03-01:
> The selector should be something like `div.inline_thumbnail > img` I would think.

could it be possible to add an "align" setting? I will place the thumbnail at the right side of file list but themify does not handle my change well. only one time when I set it up it is in the right position. after refresh it is in the old position.