jneilliii/OctoPrint-UltimakerFormatPackage #4: Static content not found - 404

OllisGit, 2019-11-15:
Hi @jneilliii ,

I installed the 0.1.1 Version on a OctoPrint 1.3.12. After restarting and opening the browser I received two 404 errors:

```
GET http://localhost:5000/plugin/UltimakerFormatPackage/static/css/UltimakerFormatPackage.css net::ERR_ABORTED 404 (Not Found)
GET http://localhost:5000/plugin/UltimakerFormatPackage/static/js/UltimakerFormatPackage.js net::ERR_ABORTED 404 (Not Found)
```

The reason is that the added ```route_hook``` for the thumbnail could not handle the static content request anymore.
I did the following changes to get the plugin to works:
```
Adjust route_hook:
  (r"thumbnail/(.*)", LargeResponseHandler....

In Javascript:
  var thumbnail_url = '/plugin/UltimakerFormatPackage/thumbnail/' + data.name.replace('.ufp.gcode','.png');
```

Question: Why is this nice plugin not listed in Plugin-Repository?
I would like to reuse the plugin in my ```PrintJobHistory-Plugin``` (https://community.octoprint.org/t/wip-new-print-history-plugin/1633/23) . 
Instead of taking a picture from the Cam it should use the thumbnail.

Thx for your work!
Olli

jneilliii, 2019-11-15:
Thanks for the fix @OllisGit.  I'll get that adjusted this weekend and work on publishing to the Plugin Repo.  It wasn't published there because I spun this one up real quick on a whim from one of the community forum posts.

jneilliii, 2019-11-15:
Hey @OllisGit, was curious if you are seeing this behavior because you have the dev option for bundling disabled, which is true by default?

```
devel:
  webassets:
    bundle: false
```

jneilliii, 2019-11-16:
I've just updated the code in the 0.1.1 branch and while testing came across a weird potential bug as I was using the Cura plugin to send the file.  All my previous testing the ufp filename did not contain spaces so this never showed up before, but when I uploaded a file it isn't finding the thumbnail because the spaces in the filename were replaced with underscores.  Have you seen this same trait in your testing? If so, I'd need to fix that before publishing to the plugin repo.

jneilliii, 2019-11-16:
Alright, I've got all the path related issues fixed and you can upload to sub-folders now as well as use the OctoPrint Connect plugin in Cura to send files with spaces in the name and it all seems to be working as expected now.  If you don't mind testing it out that would be great.

https://github.com/jneilliii/OctoPrint-UltimakerFormatPackage/archive/0.1.1.zip

jneilliii, 2019-11-16:
Just made it official 0.1.1 release and submitted to plugin repo.

closed 2019-11-16 by jneilliii

OllisGit, 2019-11-16:
@jneilliii great!!! Nightshift?!?!

- Yes, I use ```bundle: false``` and ```minify: false```
- Files with spaces works as well