jneilliii/OctoPrint-UltimakerFormatPackage #7: Plugin causes bug in gcode editor plugin

frezap, 2019-11-20:
Unfortunately the plugin seems to interfere with how the gcode editor plugin determines file names.

This is how it should look like:
[image]

This is how it looks with inline images enabled:
[image]

jneilliii, 2019-11-21:
Where is that gcode editor plugin? Does the same issue occur if you install the previous release?

frezap, 2019-11-21:
It is here: https://plugins.octoprint.org/plugins/GcodeEditor/

This was already a problem in 1.1.

jneilliii, 2019-11-21:
Yeah, after looking at that code @ieatacid took an interesting approach to get his plugin to work.  It's using post load jquery calls to inject the button and acquire the filename/header info whereas I'm injecting the button and image to the template itself and using standard knockout observable calls to get the name of the gcode, etc. 

What is causing the issue is in the line [here](https://github.com/ieatacid/OctoPrint-GcodeEditor/blob/e49e954971df21f41db1f5c3889743fa93c87a74/octoprint_GcodeEditor/static/js/GcodeEditor.js#L206) where he's walking up the DOM tree to get the name to display at the top of his modal popup.  If he were to change that to a more specific jquery selector it would be a better approach to avoid conflicts with any other plugins that may be doing manipulation to the file list the way I am.  Something like below should resolve the issue, and I'll verify by forking his plugin and making the change.

```
button.parent().parent().children("div.title").eq(0).html()
```

jneilliii, 2019-11-21:
I can confirm that change does resolve the conflict.  I will submit a pull request to the other plugin's repo with the fix.  You can install the fixed version using the URL below in plugin manager.

https://github.com/jneilliii/OctoPrint-GcodeEditor/archive/master.zip

closed 2019-11-21 by jneilliii

ieatacid, 2019-11-21:
@jneilliii, if you submit a pull request I'll merge it. I haven't had much time since my son was born.

jneilliii, 2019-11-21:
No worries, I just did.