jneilliii/OctoPrint-UltimakerFormatPackage #10: Req: Thumbnail after "DelteAfterPrint" Plugin move

TAOGde, 2020-03-01:
Hey jneilliii,

thx for this plugin, really helpfull :)
after moving the gcode file inside of OctoPrint the thumbnail disappears.
The plugin [https://plugins.octoprint.org/plugins/DeleteAfterPrint/](url) i will use with thumbnails inside the moving drawer.

any chance to fix this?

jneilliii, 2020-03-01:
I'll have to add a check for this other plugin, but could you share your settings for that plugin so that I can see exactly what you mean?

TAOGde, 2020-03-01:
> I'll have to add a check for this other plugin, but could you share your settings for that plugin so that I can see exactly what you mean?

.row-fluid inline_thumbnail > img. (the cuted entry)
[image]

TAOGde, 2020-03-02:
> I'll have to add a check for this other plugin, but could you share your settings for that plugin so that I can see exactly what you mean?

So, that's what it looks after finished print.
[image]

these are my settings
[image]
[image]

jneilliii, 2020-03-02:
Perfect, thanks. I thought you might be using the move option. I'll use that setting as my check and react accordingly.

jneilliii, 2020-03-22:
@OllisGit, while posting a request to OctoPrint's issue tracker for a FileMoved event to process differently than FileRemoved event I think I might know a workaround for this issue in my plugin.

I could create a delayed timer on FileRemoved event to delay the image removal after the FileRemoved event is triggered that gets cancelled on the FileAdded event if the flagged name is the same and then moves the image instead. Timing on that would need to be verified though, is there ever an overlap of the two existing event triggers? Only other issue I can think of related to that delayed timer is if a file gets overwrote because of an upload with the same name. Does the FileRemoved event get triggered in those situations as well or just the FIleAdded event?

In either case, I've made the request for a FileMoved event [here](https://github.com/foosel/OctoPrint/issues/3491).

jneilliii, 2020-03-23:
I just pushed an update to the 0.1.4 branch that should resolve this temporarily using a 5 second timed delay on the delete until the FileMoved event gets implemented in the future maybe. My testing with the FileManager plugin (cutting and pasting between folders) seems to work as expected. @OllisGit uses the same functions in his plugin. I was getting some random analysis errors while testing, so not sure if that is related to me trying to move the file before MetaDataAnalysis was completed yet or not. If you'd like to test you can use the URL below in Plugin Manager and report back @TAOGde.

https://github.com/jneilliii/OctoPrint-UltimakerFormatPackage/archive/0.1.4.zip

OllisGit, 2020-03-25:
@jneilliii tested the latest 0.14 version and everything is working as expected.

Thx
Olli

PR #12 (2020-03-29): 0.1.4
Added alignment option for inline images, resolves #11 
Added additional file removals to reduce disk consumption, resolves #13 
Added delayed deletion of files to avoid removing thumbnails for files moved, resolves #10
closed 2020-03-29 by jneilliii