jneilliii/OctoPrint-UltimakerFormatPackage #11: Feature Request: Add Align Setting

jneilliii, 2020-03-01:
> The selector should be something like `div.inline_thumbnail > img` I would think.

could it be possible to add an "align" setting? I will place the thumbnail at the right side of file list but themify does not handle my change well. only one time when I set it up it is in the right position. after refresh it is in the old position.

_Originally posted by @TAOGde in https://github.com/jneilliii/OctoPrint-UltimakerFormatPackage/issues/3#issuecomment-593071209_

TAOGde, 2020-03-01:
well this does noting to the image here.
align or width not working.

jneilliii, 2020-03-01:
@TaoGde, I was just enabling Themify to test.  Did you try float?

jneilliii, 2020-03-01:
That doesn't seem to work either.  I think it's because Themify doesn't apply the css live, so newly added DOM elements (async file list for example) aren't getting the setting added to Themify. I'll add a setting to apply the float option to handle alignment, shouldn't be too difficult.

TAOGde, 2020-03-01:
> @TAOGde, I was just enabling Themify to test. Did you try float?

i tryed but nothing changed. I post2 screenshots with this part.
the second screenshot is with hardcoded align="right" before your width="30%" entry.
[image]
[image]

jneilliii, 2020-03-01:
Ok, I just pushed a new branch, 0.1.4.  You can install from the URL below in plugin manager.  Give it a try and let me know how the alignment option works for you.

TAOGde, 2020-03-01:
> Ok, I just pushed a new branch, 0.1.4. You can install from the URL below in plugin manager. Give it a try and let me know how the alignment option works for you.
> 
> https://github.com/jneilliii/OctoPrint-UltimakerFormatPackage/archive/0.1.4.zip

this works :) TNX

i will try to set the thumbnail as background so the text can float on the left space ;) I give report back :)

jneilliii, 2020-03-01:
That unfortunately probably won't work because the image is contained within a div that has a class of `row-fluid` which is blocking and set for 100% width..

PR #12 (2020-03-29): 0.1.4
Added alignment option for inline images, resolves #11 
Added additional file removals to reduce disk consumption, resolves #13 
Added delayed deletion of files to avoid removing thumbnails for files moved, resolves #10
closed 2020-03-29 by jneilliii