jneilliii/OctoPrint-UltimakerFormatPackage #13: Uploaded UFTP file not deleted from Plugin-Folder

OllisGit, 2020-03-21:
Hi @jneilliii ,
I am starting analysing the "[remove thumbnail-issue](https://github.com/OllisGit/OctoPrint-DeleteAfterPrint/issues/11)".
So, I tested a "normal" case:
- Upload the ufp-file
- The ufp file is uploaded to  "data/UltimakerFormatPackage/"
- Its extract to the data folder and the gcode-file is copied to uploads/folder

fine....but after deleting the gcode from the GUI, all files where delete, except the orig. ufp file.

Tested with: 1.4.0rc5, Python 2.7

Could you reproduce this behaviour?

After upload:
[image]

After delete:
[image]

jneilliii, 2020-03-21:
No particular reason specifically that I remember, just an oversight.

jneilliii, 2020-03-21:
Just an oversight @OllisGit. It was my assumption that the [DiskFileWrapper](https://docs.octoprint.org/en/master/modules/filemanager.html#octoprint.filemanager.util.DiskFileWrapper) based on the docs moved the file instead of copy the file.

jneilliii, 2020-03-21:
Should be fixed with commit https://github.com/jneilliii/OctoPrint-UltimakerFormatPackage/commit/a66166bd089fb8709df804d6c8a82d6dc8f77f3c and added to next release version #12

OllisGit, 2020-03-22:
Morning @jneilliii ,

I tested the 0.1.4 branch. 
- Removing the file via UI is now removing completely all files
- But why do you still keep the ufp.gcode file in the plugin folder. It is redundant and not needed (IMHO)

I tested also the [FileManager](https://plugins.octoprint.org/plugins/filemanager/) plugin for moving gcode to a sub-folder. 
Unfortunately only the remove-event is passing the UFP-Plugin, the "FileAdded" event doesn't occourse....I think to solve the move image-issue (a.k.a. additional metadata), we need to analyse/improve the Octoprint-Core code or patching all plugins which were using file move.

jneilliii, 2020-03-22:
Again, I thought the removal of the source file was handled properly by the DiskWrapper [here](https://github.com/jneilliii/OctoPrint-UltimakerFormatPackage/blob/a66166bd089fb8709df804d6c8a82d6dc8f77f3c/octoprint_ultimakerformatpackage/__init__.py#L101), but I guess not. I'll have to add the manual removal. I didn't even think about the FileManager doing this too since I don't use it.

jneilliii, 2020-03-22:
Just pushed another commit that also removes the gcode file from the plugin's data folder.

PR #12 (2020-03-29): 0.1.4
Added alignment option for inline images, resolves #11 
Added additional file removals to reduce disk consumption, resolves #13 
Added delayed deletion of files to avoid removing thumbnails for files moved, resolves #10
closed 2020-03-29 by jneilliii