jneilliii/OctoPrint-UltimakerFormatPackage #14: Possible to extend this plugin to include PrusaSlicer Thumbnails?

TechnoSwiss, 2020-04-20:
PrusaSlicer is now embedding a thumbnail into the gcode for the PrusaMini, but it's also possible to setup your printer profile to include a thumbnail for any printer (supposedly, I haven't finished testing this yet).

Or maybe it's better to just create a new plugin that provides the same model preview but reads the thumbnail from the gcode file (since it's not doing any compression like the UFP)

Here's what the start of the PrusaSlicer gcode looks like with the embedded thumbnail
`; generated by PrusaSlicer 2.2.0-alpha3+184-win64 on 2020-02-04 at 09:59:07 UTC

;
; thumbnail begin 16x16 928
; iVBORw0KGgoAAAANSUhEUgAAABAAAAAQCAYAAAAf8/9hAAACfUlEQVR4AWWSWU8TYRSG52cYo2JToM
; oqtSnFNtBl2pnpOp3O1g7DIKhECKjEC3EJIpgohkhi0GBdEi6MQQk3xsQYNW43/qnX73wwdeHineWc
; 8z7nnG9GSKVSIKXTaa5MJtOSLMtQVRWSJLVifp3vE/yEKIrIZrNcuVwO+XweiqJwc7FY5KI45anW9w
; m+gQpJ1LVRUTCrJmGXJA4hUaxcLregPkzwTZTwO12upbGp9+K5E8H7iwn8nD3DdcMY5itVKpUWSKBR
; yaRpGu9Auu+KuDJ8HBtad8tMumUmUSqVOITqC4UChGq1CsMwOJUSpAdeFq9GT+HDheg/gC0vhrujIt
; bGJd6IIEKtVgOJXkj0fM1IYavejxdWH5pGL5/ksdaDJakDi9l2rCid2LAimLP3J5i0qzhX1zA/WsGC
; o2DJSGAu0YarIwHcy4f4M2l3Ioofl0aw7UXwjIFXzRiETTeO3fEw3k1EsO0O4GH5JDaq3dzwmq3xcS
; qKm2IQs/E2fJmJ49d8Ep+m42ia/XDEQQhP3AS+zwzh23QMO14Y1zNBrDMIAd6ODfD4styBhXSQmz8z
; yGKuA0vFPui6DmHNivF9qfDO/o4kAuyMhXl8tXgC62oPnrKu5wePwhsKwtIq/PCF5lgCX1kRFb5kh/
; ZI7eKirm/YBDTdityJyegROOHDaMS7YNdUWJa1BzirKZiKHcP8cIB9+wBus/GWlRA76RCDhNhK7TD7
; D0E/HYCdT6PRaMC27T8A2kNXy9CLEpMMoySjJiWhyylUUzEY7F5XC/A8D67rtgCmae6dAV2IRERK1O
; t1XuQ4zgERgO5URx4OoJ/HhxCVkj7ob/0fp3ry/gZlVv8Ubh9O9wAAAABJRU5ErkJggg==
; thumbnail end
;

;
; thumbnail begin 220x124 18480
; iVBORw0KGgoAAAANSUhEUgAAANwAAAB8CAYAAAACRt5vAAA16klEQVR4Ae2dd58UVdr++0U8G55dUa
; LkIU3OwwSGGZicc2RyBkRFEZHgmkABATMCigqKCUXSSGYM67rPb//Y/ezuZ9/BvoLzO9epOtWnTld1
; V1VXdRj94/pUd09Ph+rzres+932CLz8/n1hVQUGBLRUWFtpSUVGRbW3YsCFsFRcXh9TGjRtJY2MjuX
; btGjl+/Lil/5H/P5Zl57uUlJSwc3Hs2DHy6aefkqqqKkv/58Zv5aSN2G2Hdtu5HYZ8boMW65DZBQWq
; rKwkExMT7Lh58+awYKrdVEQ6y/LJ1pr1ZFddNlUOeaI2m3SX58cdlOXl5aSuro5UV1eT5uZmR+d2Ns
; EXNnBegRZJyJw0Aq6ysjImNLaGhgZ2ZXfaWBs2byAHm7PI1f4U8u1gqqGu0b+9155MJsvTYxY6IwG8
; np4edo7suqVbAHoNn1vg+WIZtEgDxlVaWkra29vJa6+9Rq5cucLuO22M9RS0vY25ZNoEMiNND6SS9z
; tSSFNJHoPcqSINHi5OCDH37dsXNnjhABjL4Pm8Bi0eIBMbzaZNm8jw8DC5e/cuuXXrFqmvr3fcAOFq
; 57vTLIMm66veJNK2aX1Y0LkNaU1pEamjYTFC4InqPKbqkkL2N/x/b28vuXTpErl48SILNcX/nU3wOQ
; XPFwugRRKyYI0JToY+Gq7Qe/futQSVWcOto7Cd7Mw0BOnL7kRyuHI5eXrDIrJv48PktdqVDC6j517o
; WEcqNhZGBLZqBtMG0ltRSKZoH/Op+hxysCWbvE2/x8muLHK5L4XcGE4jN4bSyM3hdHJrRNG39P7hxl
; TtHCK03LNnD3n22WdDnr9Yhc8r8HzxBJrbgMnONjAwQLZu3WoLLDNN1eSzhikDdISC1psyh2k44yEy
; mT2XbM2dR3bkzSevVi03hG5PTVrYMNVSkBDe9lQUkG21+WR3fS45RGF6sz2DnOnJJldo//E6fS8OkZ
; luy4+p4OF/t21KCjhnSDZ1dnZa/h0iAV80wfNFGrRYggxClm1oaIj12dD5d8stTnZlB4BzsSeJ9Kc9
; aArck4ULyYXOdQH/937LauYcZoKbNpQVky1VRQpMDbmKM3Vkko9oSAuY0IeEO4UCyqnw+n8qTwgIT3
; GsqKhg5xmy89vEEnxugeeLRdC8how7WldXFzl9+jT5+eefSXd3t6vh2eWBQHf7sG1NSODOtqwJ+L9P
; aFi5vTKLbKvLJ881ZpJX2nJYiHe+J51c7k/xFCQ7wL24eanp+ejo6GB94pdeesn2bxUJ+CIFni9eQX
; Pyo4mhYVtbG7l69Sq5fPkyqampcQRVMNe5PJhhGB7uolCZAYf+3CWDvtwH1OEu0f5TtKGyAtzm4gLD
; c4UMJkoHn332Gfn4449ZTbOkxHkmNV7B80UbtEhCxsMbFGlra2sZdOGCZaQa2l8605pIbo9mGIaVB0
; oWBwC3u2gR+doAtusDKazf9/WW5LgArrIwJ+h5BGiIJnCRk0P4WILPK/B8boIWK25m9GMjhISamprI
; 5OQku+K6AZeosao88g7N6F0fTGNudXcyzzBxwjKVFLyP2taSU42ryMfta01LA+jTTWbPixvgagvSde
; fEDD6c/6mpKdZ3NntOPLmeVY58kQQt0pBxAbKnnnqKnDlzJgC0cAATtYtm/W4KDRCh49GqFeT+FH18
; yFkt7mJPItlb/DDr98ULcA15iUHPk3jecQHEOMzdu3eT/v7+oL9hvLieY+CiAZqbkHFXQ/hy7tw58o
; 9//IP9qG4Bxh0TqttcTK4Mpmsp8tsqcNDB8mVkejiD3BvLtD7ShCVK1rI63QCFbTYBJwshPkoxP/30
; ExsMbSXE9wq+SIDnizfQrPwgYn/h1VdfJR988AH7Yd2Ay0gA7jqFijW+Yb/DQU8ULGTgnKSh4/RQOr
; lD+3V36N/hejeYUjWhv/Z5VyKrx41nzWWguQ7csP8zxgJwEKIOZDFPnTpF3n33Xcu/sV34YgE8nxew
; uQ2anR+AJ0ZQbEXaH0C4CVdI4FSHg7Nx4B7PX0AeoVnIx/MXkuc2LSGv1ayktbU15Dztu33YuoacaV
; 7FIEPiZCTzIdJDEypbUudowG3Nmec6IF4B17g+UXdu7MKHRApLPDnIHEcDPLvQ+SINmhduJjoa+msY
; /4grJv/x3AYsFHBwkM+7EwOA25YLzWMATVJN0AzlaOZclq0EXFsoaN3JDwQAd7R6RdwCJ8sKdBgahj
; ATvyfcT+z7RcP13ATPF2ug2YWMq6WlhTz33HPkz3/+MxkfH/cUMIy3FFVftlEPnKoXaAMMF7ipOHA3
; O8DZAXBwcJBMT0+zrHKpQeIl0q7nBni+eAcNgpudP3+e/P3vfyd9fX2uQibDZSQz4N5uSGCQOQWuL/
; VB8mzpklkNXDD4MNtg27Zt5IcffiDvvPMOO9fxDp7PCWzRAM3oCogfACEHCtgYnc7Dj3AgswKYVeDO
; 0v7ZTupwYzQBggK3VeB6BYf706Z4Ay5Jd27ChY//vidOnGChppkbxiJ4IYGLNdBC9b22bNlCRkdHtS
; tfpACT1VBeEgS4BeQx1eEUiOaQwfQHGWwAEcChn9ZH/9aV9ADpSFSA61OBQy0vnoGTFa7z8WJ5sLYR
; y+D53AofvXYzLmSvMEH0iSeeYNlIuz9iuHDxZRdEBQduoQbclOpqSPkDOg5aT/IcFkZ2UuC4w/mBWz
; 6rgAsXQH7OkYEO5nhuu55bYaYvEq7mBmg42QgtXn75ZfLvf/+bPP30055DZgSXHeA+pTU1M+DGqLON
; 0FAS4KGv1p1sDBzqd/EEXFN+kuPzbQc6TBD+8ccfyWOPPRa03bgJnhtu54t10ERne//999lUGl5fcx
; syq4DZA25BWMC9E4fAyefHC/iQUMHk1jt37rARKlbaUCyA54tm+GjlJGGsHepqOMm4HSuQcaHW11hR
; aggcliQAcDxLuTUocA/MWuC8gg+vh0mtvH2ISbNI9PGcQOeLVdBw8tBHw5UM4QOObkAWLlxGMgWu3w
; /cDofAnetYN6uACxdAM/gw3erxxx83TLTEEng+t2BzCzScLFy1EJt/8cUXbMBxuKC5DZhV4KCdauE7
; ALgsY+B64h64ZMfn2w3w4HYYj4n+fiTBswqdb9euXezLxgJoPDGCAcf/+te/yDPPPBNRyKwCZg84we
; FyQwPndzjl8XPt6zwZbOwlcPL5iSR8yFweOHCAjTgSE2uxAB7atg+NGmPWvIDNDmhcGCmCUSOtra22
; QYskZFwIe5sqN5kCt3fjw7aAg8Mpo0zmGA/rilH4ggEXKfjEhMrIyAhbzgHLOojtyw3w7EKHx/C6GK
; rmwx00GoxX4+BFGjScLICGIrYTR4sUYDhPRgoG3KvVKxTg1gcCN2wI3BwNuJONq9nMg9tmDT2G4LMK
[... 180 lines cut ...]
; GnSLfxMzkgEuFlyXA80FFy5nUhogjpDfEWijlBouD8nweMUKvyrZ//tfb9PS3zLhffhRe18u4flcFc
; v1ryu/hxOJr4H3AGiNFaUhQYs2cLN6eo4ZdGZfDF+6oiBLgu5/hPv+22YqNjgWG/y/2fOKgzzmCNgg
; zhtMJUFUulQRA3CZos3LDABcrpcCoATjCv+RyQjO5f5juaSaLFpLqyghLS0tlmHzuv8WCeDs9N/CAs
; 6LxIkMXVVJYUiw4lmW4A0CqRF8par7hYTQAMRAVzS6rTyH9aOpm1Vn0T6aChpXrLtbLGYoYwI4qKai
; jJSuWxiXQEUCVkMnXBocQCMIDYE0UcVqBbSmumodaJF2t1kLXDT6cfKJKMtOiisgIguf3w0DHHCpDK
; AEoQRiMBjLKWi1+WmGoMUabPHWf/N8IdhQwBlBB7fbuPKBuIIhsuAJYapZ+GkI4G9MARRBM4LMCWzx
; 5m6R6L9FDDg7LqedmMpyUp6TxLKZXKXrFmkClL90MHX9wyDwGTofIKP9s+09TeToCwfIoUOHSF9fn2
; vAuQFbvIWTrgAXqX6ckYKdULMfoQ67qVRXMNVsKtZUVZTDVJGbTJWiqSw9gWlT4iKdNq6cwxRX4IWA
; bvOqB0lF8mJSQ50M9TMO0oEDB8iFCxfIvn37SGdnZ0Rgi4a7Rbv/FgBcNPpxwaBzAhxk5cpppUHIDa
; mhpkqnurKNTNUbcjVV5qUy1RRkkNq8JFJJU+nVOevI5qTFFOSHmUoS5mhyDzp9HXFTwoOkjL5nZeZq
; UrcxTweYkXp7e5nDQUNDQ6S1tdVRGBmP7hap/psnwMWCy1mBzupV2G5jQyNta2tjjXZsbIx0dXUFbe
; iiGmurNdVTQLhqivOYOMyi+N8gPLepttry+xlpfHycDAwMsO8A6LyAbTa6m6vAxVtYGS3o0FBPnz5N
; XnzxReYY4TT8aGvnzp3klVdeIbt372bwxQps8RxOQv8fEXOcTcmPBu8AAAAASUVORK5CYII=
; thumbnail end
;
; 

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; first layer extrusion width = 0.42mm

M73 P0 R122`

jneilliii, 2020-04-20:
I would probably do this as a separate plugin since I'll have to translate this encoded string into a file on upload instead of extracting the file from a zip container. Please confirm this is possible with any printer profile and let me know how so I can verify from my own local installation.

TechnoSwiss, 2020-04-20:
I can confirm this works on any printer with PrusaSlicer. Has to be done in a text editor, no way to do this from the GUI, but if you open up the definition file for your printer and locate the line 'thumbnail = ' and change to something like 'thumbnails = 16x16,220x124' re-open PrusaSlicer and it will now include those 2 thumbnail sizes in the gcode.

TechnoSwiss, 2020-04-20:
Looks like from this closed issue on the PrusaSlicer project they're waiting to see OctoPrint support embedded thumbnails before they include embedded thumbnails on more printer profiles.

https://github.com/prusa3d/PrusaSlicer/issues/3488

jneilliii, 2020-04-20:
Yep, so it looks like this wouldn't be too difficult, but will require some work to make the thumbnail extract-able from the gocde since the resolution is configurable there and for example you're providing two in your example above. It's base64 encoded string and I would just have to either use it as the image url like below or convert it on the python side to an actual png file.
```
<img src="data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAANwAAAB8CAYAAAACRt5vAAA16klEQVR4Ae2dd58UVdr++0U8G55dUa
LkIU3OwwSGGZicc2RyBkRFEZHgmkABATMCigqKCUXSSGYM67rPb//Y/ezuZ9/BvoLzO9epOtWnTld1
V1VXdRj94/pUd09Ph+rzres+932CLz8/n1hVQUGBLRUWFtpSUVGRbW3YsCFsFRcXh9TGjRtJY2MjuX
btGjl+/Lil/5H/P5Zl57uUlJSwc3Hs2DHy6aefkqqqKkv/58Zv5aSN2G2Hdtu5HYZ8boMW65DZBQWq
rKwkExMT7Lh58+awYKrdVEQ6y/LJ1pr1ZFddNlUOeaI2m3SX58cdlOXl5aSuro5UV1eT5uZmR+d2Ns
EXNnBegRZJyJw0Aq6ysjImNLaGhgZ2ZXfaWBs2byAHm7PI1f4U8u1gqqGu0b+9155MJsvTYxY6IwG8
np4edo7suqVbAHoNn1vg+WIZtEgDxlVaWkra29vJa6+9Rq5cucLuO22M9RS0vY25ZNoEMiNND6SS9z
tSSFNJHoPcqSINHi5OCDH37dsXNnjhABjL4Pm8Bi0eIBMbzaZNm8jw8DC5e/cuuXXrFqmvr3fcAOFq
57vTLIMm66veJNK2aX1Y0LkNaU1pEamjYTFC4InqPKbqkkL2N/x/b28vuXTpErl48SILNcX/nU3wOQ
XPFwugRRKyYI0JToY+Gq7Qe/futQSVWcOto7Cd7Mw0BOnL7kRyuHI5eXrDIrJv48PktdqVDC6j517o
WEcqNhZGBLZqBtMG0ltRSKZoH/Op+hxysCWbvE2/x8muLHK5L4XcGE4jN4bSyM3hdHJrRNG39P7hxl
TtHCK03LNnD3n22WdDnr9Yhc8r8HzxBJrbgMnONjAwQLZu3WoLLDNN1eSzhikDdISC1psyh2k44yEy
mT2XbM2dR3bkzSevVi03hG5PTVrYMNVSkBDe9lQUkG21+WR3fS45RGF6sz2DnOnJJldo//E6fS8OkZ
luy4+p4OF/t21KCjhnSDZ1dnZa/h0iAV80wfNFGrRYggxClm1oaIj12dD5d8stTnZlB4BzsSeJ9Kc9
aArck4ULyYXOdQH/937LauYcZoKbNpQVky1VRQpMDbmKM3Vkko9oSAuY0IeEO4UCyqnw+n8qTwgIT3
GsqKhg5xmy89vEEnxugeeLRdC8how7WldXFzl9+jT5+eefSXd3t6vh2eWBQHf7sG1NSODOtqwJ+L9P
aFi5vTKLbKvLJ881ZpJX2nJYiHe+J51c7k/xFCQ7wL24eanp+ejo6GB94pdeesn2bxUJ+CIFni9eQX
Pyo4mhYVtbG7l69Sq5fPkyqampcQRVMNe5PJhhGB7uolCZAYf+3CWDvtwH1OEu0f5TtKGyAtzm4gLD
c4UMJkoHn332Gfn4449ZTbOkxHkmNV7B80UbtEhCxsMbFGlra2sZdOGCZaQa2l8605pIbo9mGIaVB0
oWBwC3u2gR+doAtusDKazf9/WW5LgArrIwJ+h5BGiIJnCRk0P4WILPK/B8boIWK25m9GMjhISamprI
5OQku+K6AZeosao88g7N6F0fTGNudXcyzzBxwjKVFLyP2taSU42ryMfta01LA+jTTWbPixvgagvSde
fEDD6c/6mpKdZ3NntOPLmeVY58kQQt0pBxAbKnnnqKnDlzJgC0cAATtYtm/W4KDRCh49GqFeT+FH18
yFkt7mJPItlb/DDr98ULcA15iUHPk3jecQHEOMzdu3eT/v7+oL9hvLieY+CiAZqbkHFXQ/hy7tw58o
9//IP9qG4Bxh0TqttcTK4Mpmsp8tsqcNDB8mVkejiD3BvLtD7ShCVK1rI63QCFbTYBJwshPkoxP/30
ExsMbSXE9wq+SIDnizfQrPwgYn/h1VdfJR988AH7Yd2Ay0gA7jqFijW+Yb/DQU8ULGTgnKSh4/RQOr
lD+3V36N/hejeYUjWhv/Z5VyKrx41nzWWguQ7csP8zxgJwEKIOZDFPnTpF3n33Xcu/sV34YgE8nxew
uQ2anR+AJ0ZQbEXaH0C4CVdI4FSHg7Nx4B7PX0AeoVnIx/MXkuc2LSGv1ayktbU15Dztu33YuoacaV
7FIEPiZCTzIdJDEypbUudowG3Nmec6IF4B17g+UXdu7MKHRApLPDnIHEcDPLvQ+SINmhduJjoa+msY
/4grJv/x3AYsFHBwkM+7EwOA25YLzWMATVJN0AzlaOZclq0EXFsoaN3JDwQAd7R6RdwCJ8sKdBgahj
ATvyfcT+z7RcP13ATPF2ug2YWMq6WlhTz33HPkz3/+MxkfH/cUMIy3FFVftlEPnKoXaAMMF7ipOHA3
O8DZAXBwcJBMT0+zrHKpQeIl0q7nBni+eAcNgpudP3+e/P3vfyd9fX2uQibDZSQz4N5uSGCQOQWuL/
VB8mzpklkNXDD4MNtg27Zt5IcffiDvvPMOO9fxDp7PCWzRAM3oCogfACEHCtgYnc7Dj3AgswKYVeDO
0v7ZTupwYzQBggK3VeB6BYf706Z4Ay5Jd27ChY//vidOnGChppkbxiJ4IYGLNdBC9b22bNlCRkdHtS
tfpACT1VBeEgS4BeQx1eEUiOaQwfQHGWwAEcChn9ZH/9aV9ADpSFSA61OBQy0vnoGTFa7z8WJ5sLYR
y+D53AofvXYzLmSvMEH0iSeeYNlIuz9iuHDxZRdEBQduoQbclOpqSPkDOg5aT/IcFkZ2UuC4w/mBWz
6rgAsXQH7OkYEO5nhuu55bYaYvEq7mBmg42QgtXn75ZfLvf/+bPP30055DZgSXHeA+pTU1M+DGqLON
0FAS4KGv1p1sDBzqd/EEXFN+kuPzbQc6TBD+8ccfyWOPPRa03bgJnhtu54t10ERne//999lUGl5fcx
syq4DZA25BWMC9E4fAyefHC/iQUMHk1jt37rARKlbaUCyA54tm+GjlJGGsHepqOMm4HSuQcaHW11hR
aggcliQAcDxLuTUocA/MWuC8gg+vh0mtvH2ISbNI9PGcQOeLVdBw8tBHw5UM4QOObkAWLlxGMgWu3w
/cDofAnetYN6uACxdAM/gw3erxxx83TLTEEng+t2BzCzScLFy1EJt/8cUXbMBxuKC5DZhV4KCdauE7
ALgsY+B64h64ZMfn2w3w4HYYj4n+fiTBswqdb9euXezLxgJoPDGCAcf/+te/yDPPPBNRyKwCZg84we
FyQwPndzjl8XPt6zwZbOwlcPL5iSR8yFweOHCAjTgSE2uxAB7atg+NGmPWvIDNDmhcGCmCUSOtra22
QYskZFwIe5sqN5kCt3fjw7aAg8Mpo0zmGA/rilH4ggEXKfjEhMrIyAhbzgHLOojtyw3w7EKHx/C6GK
rmwx00GoxX4+BFGjScLICGIrYTR4sUYDhPRgoG3KvVKxTg1gcCN2wI3BwNuJONq9nMg9tmDT2G4LMK
XDgA2gGPPx/vgWwmuinRAA9HDKgHbHhfHx7kSQrcRgxsttqw26ChscLJMBYSnV6k/q2CFgnIzACzAx
wSJ06A040wGfYfb8cYaDJwzQXJjs+32/DxNoY1bDiEckTlJXjIPyCMxGvg84Avn2h7sOKzZ8+y7ZjG
xsZ0LxYubPIXRayN/iMGHB89ejQm3MwqYHaAg1CPMwIOc99k4PC3I8FGlwjwaQDKf4sicP1F69j5EM
9PLICH10Vu4LvvvmOrANgFzw50MA7UBjGh9tFHH2X/zznzySEk3O2NN95gT8a4tVBvZhc0nkl66623
yD//+U+WynUTtEhBBiEEZ3PwqoIDB31Dne6NugSyr+ThAODQV8P9VyqXGwIWLKS8rR0zoup+HLhjtQ
nk3kQOuTOeTW6OZpGvhnPJhf5scrI7lzzfsYEcaC0kw3UlpKO8yFX4rLQhXOSRSEFCBcvZG7XNcMAD
PwhdYSBgiE8rEkNMXzA7xD/jS8P53IANr8X3WEOnFrGtG6BFCjIOmCwrwInCGpPQx53r6NH9tUoAnw
il7hgB4O5P5oYUoLw1lk0+H8gmRzrzyf7WAjJQUxw2fKHaEwZPYMMRRFdo33KW0yl0eBybVKI9IIzk
W3fJ8oVKjMDlEAOH62r4Ymi0L7zwAuuvRRo0NwELFzhPJa9ZIjqkh/BZAW4mxGO4fZtCeKEvizzXXm
QKoFuOB/CefPJJ05qeHfAQRmJZD0AXrF/ns5oYgeNhqblgUyOMPjQaJDI0R44cYf8bSdC8AiyWgRPD
z5sqbLpMpwGQkQBuxuJjsm7QsPTDngyyv6XAM/BgADt37mRtwC54mLVy+PBhjYtQCRWf1SwkgHvvvf
fI66+/ztabwAcNBRtCRtT5/vOf/7A1IWMRNKeQQQhPoOaqzbHjcBZhDFCYq3lx4J6jujSQTq4OZTBY
7k7kkh+25TPNTOVZCjfNdHc8h3xK3U+GL1zw8Hf0ub7//nvTkU0yaDwPgXaNNm7W7ZKh81mBjb8JYl
5kXw4dOsTs0ww0noJFOPq3v/2NHb0GLRKQccBkxQtwMny3zaBzAB8HDokgJIEwjE1ZqWwBeapoEXme
/u04db/TLWvJV/3p5BoFEgB+DxBtQDej6g4NPT/oziCP1ea4Ah4Aglth9gEy9MHCTBx37NhBnn/+eb
Z+Dn/cShbT5yTdj3gVwoxr2Ybx4VFXw9/QSPkk0XgEzQyw2QIcjjcNdt65OWySeLEA3CgFbjstf/Dl
JMRlJURxGLFs4Bv1q8jnW9LIbepgP2zPD9rHmzF47KuhXPIszX6GCx5vI0h4oA0bzU7B4zAb/j52yw
c+J+l+3jfjhWqx8eE+YmJ8MKeu5iZoXkEW78DdMnA3Eb6bwirSt0T3CwHcGAUOMEHboVzhqEHo11bh
9qO0VrmvZDE5QZ0QLniHAvj91vWmsMlCGQKlB4xttQJesDATmUYMyuBZTCRD4IKAEZK7VFbB8zmBTe
6nIZbFCG3si43iYryB5gQyCHE71FJdFv/ASfDJcAVkOs2Ao/VEFPkVKSNsMI50hwqhBqMKGQdRJxVW
[... cut ...]
5m6R6L9FDDg7LqedmMpyUp6TxLKZXKXrFmkClL90MHX9wyDwGTofIKP9s+09TeToCwfIoUOHSF9fn2
vAuQFbvIWTrgAXqX6ckYKdULMfoQ67qVRXMNVsKtZUVZTDVJGbTJWiqSw9gWlT4iKdNq6cwxRX4IWA
bvOqB0lF8mJSQ50M9TMO0oEDB8iFCxfIvn37SGdnZ0Rgi4a7Rbv/FgBcNPpxwaBzAhxk5cpppUHIDa
mhpkqnurKNTNUbcjVV5qUy1RRkkNq8JFJJU+nVOevI5qTFFOSHmUoS5mhyDzp9HXFTwoOkjL5nZeZq
UrcxTweYkXp7e5nDQUNDQ6S1tdVRGBmP7hap/psnwMWCy1mBzupV2G5jQyNta2tjjXZsbIx0dXUFbe
iiGmurNdVTQLhqivOYOMyi+N8gPLepttry+xlpfHycDAwMsO8A6LyAbTa6m6vAxVtYGS3o0FBPnz5N
XnzxReYY4TT8aGvnzp3klVdeIbt372bwxQps8RxOQv8fEXOcTcmPBu8AAAAASUVORK5CYII=">
```

jneilliii, 2020-04-21:
Took a quick stab at it tonight. Still very Beta and needs a lot of testing. The images aren't as clean coming out of Prusa as they are with Cura because of the background (bed) in the image, but it does work.

[image]

I've posted the initial development version for this over [here](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails). Feel free to start beta testing and report issues in that repository. Right now it will process every uploaded gcode and add the button/inline thumbnail for every gcode file you have uploaded but won't extract the image from those previously uploaded, only files uploaded once the plugin is installed. 

The caveat currently that I'm too tired to dig into is that it will grab the first thumbnail it finds in your gcode, so with the example you gave it would extract the 16x16 image, not the larger one. I took your example and removed the first thumbnail for the example screenshot. If you were to switch your config to `'thumbnails = 220x124,16x16` then it will extract the bigger one. I don't really want to deal with handling multiple thumbnails for the web interface based on the dimension flags in the comments.

Without looking myself yet I was wondering if PrusaSlicer's interface for the thumbnails gives you a set of options or can you manually enter it with the mini?

closed 2020-04-21 by jneilliii