jneilliii/OctoPrint-UltimakerFormatPackage #19: How to save .ufp.gcode from cura.

leoheck, 2020-04-28:
First of all, this plugin is awesome, thank you.

Everything is working just fine using `Print with octoprint` button. My doubt is, how do I save this `.ufp.gcode` instead of printing directly with octoprint?

leoheck, 2020-04-28:
I see, you use the .upf to get the image and the gcode all at once. Clever! So I just have to save the .upf file and use it when I need to work like this.

jneilliii, 2020-04-28:
There's also the octoprint connection plugin to cura that will give you one click printing directly to octoprint and supports the ufp file format if this plugin is detected.

leoheck, 2020-04-28:
> There's also the octoprint connection plugin to cura that will give you one click printing directly to octoprint and supports the ufp file format if this plugin is detected.

Yes yes,  that is what I said in the first comment. I use this a lot at home with my printer.

But at my work, I have to store the files before sending them to print. So, right now I am storing this .ufp instead of .gcode since it is just a package. Now, I have another problem is that I upload (many files) them with ssh instead of just using octoprint website.

So, my second question is. What you are doing with this plugin so I can mimic it from the command line? You send the .ufp, then unpack it then, rename the gcode, where is the picture? Where the picture will be stored? Can you explain this to me?

leoheck, 2020-04-28:
Your plugin has any way to run it from the command line passing the .ufp and then it does the job?

leoheck, 2020-04-28:
Adding the .ufp and the .png in the `/home/pi/.octoprint/data/UltimakerFormatPackage` dind't work as I was expecting.

jneilliii, 2020-04-28:
> Your plugin has any way to run it from the command line passing the .ufp and then it does the job?

Transferring files outside of API or via the web interface (which also uses API on the backend) should be done to the watched folder. That way the "normal" file processing occurs on those by the system. This is all in theory though as I have not yet tried it. So from Cura you save your sliced file as ufp format and then SSH transfer your file into the watched folder. If that doesn't work then there may be a bug in the core OctoPrint where it doesn't take into account custom file extensions like it does in the web interface.

jneilliii, 2020-04-28:
I just tested my theory and it does work as expected on my pi, but not in windows for some reason. So yeah, slice, save your file as ufp format, upload/transfer to the watched folder.

[image]

leoheck, 2020-04-29:
Awesome, this `/home/pi/.octoprint/watched` works, thank you.

Do you know if it is possible to organize files in a folder? Adding a folder to the watched does not work.

leoheck, 2020-04-29:
Ah, maybe it is not possible since it is an open request
https://github.com/OctoPrint/OctoPrint/issues/2451

jneilliii, 2020-04-29:
Correct, it's not currently possible.

closed 2020-05-03 by leoheck