jneilliii/OctoPrint-UltimakerFormatPackage #20: Updating to 0.1.7 broke remote printing

sidhant, 2020-05-01:
After updating to 0.1.7 release, I have not been able to print from Cura. I get the following error:

jneilliii, 2020-05-01:
Version 0.1.6 and up actually have this issue with the auto-start option from Cura. It still uploads the file, it just isn't able to start it automatically. The Cura plugin will be getting an update soon that fixes this, or you can downgrade to the 0.1.5 version by using the URL below in plugin manager.

https://github.com/jneilliii/OctoPrint-UltimakerFormatPackage/archive/0.1.5.zip

I will leave this ticket open until that new version is released.

sidhant, 2020-05-01:
Thank you for the prompt response! I did downgrade to 0.1.5 and it is still giving me an issue. Its OK, until Cura plugin is updated, I will start jobs manually.

jneilliii, 2020-05-01:
You can also follow the steps at the forum post below to test the newer Cura plugin version.

https://community.octoprint.org/t/cura-octoprint-connection-plugin/5348/224?u=jneilliii

sidhant, 2020-05-03:
> You can also follow the steps at the forum post below to test the newer Cura plugin version.
> 
> https://community.octoprint.org/t/cura-octoprint-connection-plugin/5348/224?u=jneilliii

Cura 4.6.1 + UFP Plugin 0.1.7 + the dev snapshot of OctoPrint Plugin works.

jneilliii, 2020-05-03:
Thanks for confirmation. I'm sure the Cura plugin will get updated soon with an official fix.

leoheck, 2020-05-04:
I also came here to say the same with the same setup and versions. What should I do to fix it?
[image]

jneilliii, 2020-05-04:
You will have to wait for an official upgrade to the cura plugin, or install the development snapshot linked in the community forum post above.

closed 2020-07-16 by jneilliii