jneilliii/OctoPrint-UltimakerFormatPackage #23: Thumbnails broken after moving gcode files in File Manager

kimdurose, 2020-07-12:
The thumbnails do not appear when the gcode files are moved to another folder in File Manager.

jneilliii, 2020-07-12:
I thought I already fixed that issue, you're using the latest version? 

https://github.com/jneilliii/OctoPrint-UltimakerFormatPackage/issues/10

kimdurose, 2020-07-12:
After a bit of experimenting, they now work if the file is moved. However, if the file is renamed, the link to the thumbnail is broken. Lovely plug-in though!

jneilliii, 2020-07-12:
Thanks, I've updated the title.

jneilliii, 2020-07-12:
Although, I'm not sure I'll be able to resolve this because I don't believe there is a built in event for file rename and the FileManager plugin is doing that on his own.

kimdurose, 2020-07-12:
Thank you for looking in to it

Kim Durose

kimdurose, 2020-07-13:
Hi,
I can reconfirm that the preview thumbnail DOES disappear if the gcode file is moved to another folder.
It seems to stay for a while then is replaced by a generic icon.

kimdurose, 2020-07-14:
[image]

kimdurose, 2020-07-14:
[image]

jneilliii, 2020-07-14:
And the file's name stays the same during the move process? That shouldn't be happening unless the process gets broken in some way.

kimdurose, 2020-07-15:
Hi
Yes, simple cut and paste of the file from within File Manager

Kim Durose

jneilliii, 2020-07-15:
I can confirm. For some reason this has broken, at least in my testing with 1.4.1RC2 and 1.4.1RC3, which incorporates it's own move option in the standard file list. I know I had fixed this at some point and not sure why it's no longer working.

jneilliii, 2020-07-15:
Looks like this may have been a bug regression. I think I just fixed it in a separate branch. If you don't mind testing it for me prior to release that would be awesome. Just use the URL below in Plugin Manager to install and let me know if it fixes it or not. Thanks.

https://github.com/jneilliii/OctoPrint-UltimakerFormatPackage/archive/0.1.9.zip

kimdurose, 2020-07-15:
No problem I will do. Need to print some stuff later so will look in to it.
Thanks again!

Kim Durose

kimdurose, 2020-07-15:
Hi
Still broken, sorry. Thumbnails are still vanishing.

Kim Durose

kimdurose, 2020-07-15:
Three files sent from Cura. All had thumbnails. All three then moved to another folder. Two still have thumbnails, one does not.

Kim Durose

kimdurose, 2020-07-15:
> Not sure why it would work for some and not others. Were you doing multiple files at once, not even sure if that's possible. Could you go into logging section of octoprint's settings, find the plugin in the dropdown, change to debug and click the plus sign, then save. Try the process a little more and then grab octoprint.log for analysis.

Hi
Each file was sent over individually from Cura. Then all three were selected in File Manager and cut and paste in to the new folder.

Kim Durose

jneilliii, 2020-07-15:
Yeah, that could have been the reason. I ended up having to put in a delay and keep track of FileDeleted and FileAdded to handle the move as there is no FileMoved event to monitor for this operation. I will need to work on making that more resilient for multiple files being done at the same time, as I suspect doing them all at once happened to quickly and the tracking of the filename overlapped. Shouldn't be too difficult to switch that to a list of files rather than just a single file, but will take a little longer to fix. If you do just one at a time with a delay between each for now that should work as expected

kimdurose, 2020-07-15:
Hi

Just done a test cutting and pasting one at a time and that appears to have fixed it. However, a totally separate file in the top lecel of the folder structure has now lost its preview 😊

kimdurose, 2020-07-15:
False alarm. The singly uploaded file has also now lost it’s thumbnail.

jneilliii, 2020-07-15:
Could you go into logging section of octoprint's settings, find the plugin listed in the dropdown, change to debug and click the plus sign, then save. Try the process a little more and then grab octoprint.log for analysis.

jneilliii, 2020-07-16:
So I just pushed an update to that same branch that should help with multiple files moved at once. I've tested with FileManager and it appears to work, however there is some additional tweaking/optimization necessary. I did notice what you mentioned about the thumbnail going away after a brief time and from what I can tell pressing the refresh button on the default file list in the sidebar did sync things back up and reload the inline thumbnail. If you want to test you can use the same URL as before to install.

kimdurose, 2020-07-16:
Hi

Thanks for the update, however refreshing the file list does not bring back the thumbnails.

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

jneilliii, 2020-07-16:
Just to verify, you mean after installing this update and testing the refresh button doesn't work? Previous thumbnails may be lost prior to this update, and if they're gone the refresh isn't going to bring them back.

kimdurose, 2020-07-16:
Ok I’ll retest later

Kim Durose

kimdurose, 2020-07-16:
Hi

Working perfectly!!

Thank you so much!!

jneilliii, 2020-07-16:
Awesome, thanks for verifying. Like I said, I need to do a little code clean-up before release, but shouldn't take too long to do. Just need to refactor a couple of things to reduce the duplication of some stuff I did to get it to work.

kimdurose, 2020-07-16:
Great stuff, thank you 

Kim Durose

PR #24 (2020-07-16): 0.1.9
fix regression, thumbnails being broken when moving 
make thumbnail moving a little more resilient for multiple moves at once in quick succession, resolves #23
closed 2020-07-16 by jneilliii