jneilliii/OctoPrint-UltimakerFormatPackage #27: No Thumbnails

onehunkytenor, 2020-10-24:
Thanks for what I expect would be a TERRIFIC plugin... if only I could get it (or maybe Cura) to work!

I believe I've tried everything but no-go. Cura works properly with OP but no thumbnail icon appears in the file list. I suspect that Cura is not sending files over in the correct format as new files still use the .gcode extension.

Any ideas?

Thanks!

onehunkytenor, 2020-10-24:
Further tinkering lead me to change the flavour of g-code to Ultimaker 2 (.ufp) and voila!

From within Cura -> Preferences -> Configure Cura -> Printers -> Machine Settings -> G-Code Flavor - Ultimaker 2

jneilliii, 2020-10-24:
You can also install the octoprint connection plugin in cura and it will detect this plugin installed in octoprint and automatically upload the ufp format.

closed 2020-10-24 by jneilliii

onehunkytenor, 2020-10-25:
I guess I did things the wrong way around? I had Cura and the OctoPrint plugin installed and working with OctoPrint THEN installed your plugin in OctoPrint. In this case, no such discovery or format switch was made automatically. I would suggest adding something to your documentation that tells people that switching the flavour of g-code is required. Cheers and thanks for a very cool plugin!

jneilliii, 2020-10-25:
Yeah, it was a little more obvious when the plugin was named UltimakerFormatPackage, but that also made it difficult to find based on what it does.  But it does list that format as the first line in the readme...I guess you may have missed that.

https://github.com/jneilliii/OctoPrint-UltimakerFormatPackage#cura-thumbnails

onehunkytenor, 2020-10-27:
I wanted to thank you for taking the time to help me out! Some more fiddling and I think I have it licked. Looks like my Ender 5 Plus didn't quite like the ufp flavour of g-code. It started doing some very strange things when travelling, like leaving these huge strings of filament and double-printing some exterior walls. Made for totally useless prints! Changing back to Marlin in Cura seems to have done the trick. It still appears to be sending ufp files (says so in Cura when sending to OP) and your thumbnails still work) but whatever was causing the strange behaviour is gone. No more errant stringing.
Speaking of your plugin, it also seems to be a little bit finicky in that the proper thumbnail (it uses the one from the last file printed) will not appear nor will a file properly delete from the file list unless and until a browser refresh has been done. Worth it for the thumbnails but hey... thought you'd like to know!
Let me know if I can supply any info to assist in debugging stuff.
Thanks again for taking the time!
Cheers,-Shawn  

 
 

https://github.com/jneilliii/OctoPrint-UltimakerFormatPackage#cura-thumbnails

jneilliii, 2020-10-27:
> it also seems to be a little bit finicky in that the proper thumbnail (it uses the one from the last file printed) will not appear nor will a file properly delete from the file list unless and until a browser refresh has been done.

I have seen certain situations where the image sticks around in the upper state panel, but it usually fixes itself when you start a new print or refresh like you mention. Just haven't had the time to dig any deeper into the specific conditions of when it happens since it was easy enough to fix.

onehunkytenor, 2020-10-27:
Fair enough. New prints also seem to land without a thumbnail in the State panel until a browser refresh. That may account for my earlier observation about a thumbnail appearing to stick around.
Anyhow.
Thanks again!-S

onehunkytenor, 2020-10-29:
Me again...
You're the author of Bed Visualizer, correct?
Is there a trick to get ABL to actually make z-axis adjustments while printing? I never seem to see ANY a-axis adjustments!
Thanks,-Shawn
    On Tuesday, October 27, 2020, 12:12:59 a.m. EDT, SHAWN SCANLAN <sscanlan@rogers.com> wrote:  
 
  Fair enough. New prints also seem to land without a thumbnail in the State panel until a browser refresh. That may account for my earlier observation about a thumbnail appearing to stick around.
Anyhow.
Thanks again!-S

jneilliii, 2020-10-29:
Please keep discussions in their appropriate place. You could use the custom controls option in bed level visualizer to send baby stepping gcode commands.