jneilliii/OctoPrint-UltimakerFormatPackage #28: Correct URL for HomeAssistnat

Codename-11, 2020-10-31:
Since one of the recent updates, I have been unable to import the thumbnail image from Home Assistant using a template sensor to read the endpoint `<octoprint-url>/plugin/UltimakerFormatPackage/thumbnail/`.

I had previously been able to use the template `<octoprint-url>/plugin/UltimakerFormatPackage/thumbnail/{{(value_json.path)|regex_replace(find='gcode', replace='png', ignorecase=True)}}`, but this no longer works.

Unfortunately I have been unable to figure out how to adapt. Did something change that would cause that endpoint to no longer allow me to retrieve the URL for the currently loaded thumbnail? Any information pointing me in the right direction would be a great help. Thanks!

jneilliii, 2020-10-31:
Nothing I'm aware of would have changed that in any recent update. Relative to your quandary though, what is the whole of `value_json` that you are working with. I just don't know what payload you are getting that information from. The relative url to the thumbnail is actually exposed via the file api directly. I think it just depends on the context/which api you're getting the info from. The property from the file api is `thumbnail`. below is an example from the file list api request.

```
{
    "files": [{
            "date": 1603865016,
            "display": "CCR10_HMC_Top.gcode",
            "gcodeAnalysis": {
                "dimensions": {
                    "depth": 233.224,
                    "height": 193.8,
                    "width": 195.133
                },
                "estimatedPrintTime": 102622.62274090867,
                "filament": {
                    "tool0": {
                        "length": 114301.22826999471,
                        "volume": 0.0
                    }
                },
                "printingArea": {
                    "maxX": 205.233,
                    "maxY": 253.224,
                    "maxZ": 194.0,
                    "minX": 10.1,
                    "minY": 20.0,
                    "minZ": 0.2
                }
            },
            "hash": "218baea4c52f22db67d20da77159f59f45ef8086",
            "name": "CCR10_HMC_Top.gcode",
            "origin": "local",
            "path": "CCR10_HMC_Top.gcode",
            "prints": {
                "failure": 2,
                "last": {
                    "date": 1603896122.6725907,
                    "success": false
                },
                "success": 0
            },
            "refs": {
                "download": "http://192.168.0.120/downloads/files/local/CCR10_HMC_Top.gcode",
                "resource": "http://192.168.0.120/api/files/local/CCR10_HMC_Top.gcode"
            },
            "size": 118009542,
            "statistics": {
                "averagePrintTime": {},
                "lastPrintTime": {}
            },
            "thumbnail": "plugin/UltimakerFormatPackage/thumbnail/CCR10_HMC_Top.png?20201028060336",
            "thumbnail_src": "UltimakerFormatPackage",
            "type": "machinecode",
            "typePath": ["machinecode", "gcode"]
        }
    ],
    "free": 12236767232,
    "total": 15384698880
}
```

Codename-11, 2020-11-20:
Strangely enough it seems to be working again. In regards to where in getting the "value" aspect, I believe it has to do with how the value is returned through the template. I used this snippet from Reddit and unfortunately I haven't been able to decipher where that particular piece comes in. All good now though!

jneilliii, 2020-11-20:
I'd have to see how your HA template is setup, but I think you could use `<octoprint-url>/{{(value_thumbnail)}}` maybe assuming your getting the data the way I think you are. Sounds like it doesn't matter though since it started working again.

closed 2020-11-20 by jneilliii