jneilliii/OctoPrint-UltimakerFormatPackage #29: Previews no longer show when print is selected

Jon-So, 2020-11-30:
In the new octoprint release of 1.5.0 the prints when selected no longer show a preview.

jneilliii, 2020-11-30:
Thanks for your report. This is something I will be looking into as it is also effecting the PrusaSlicerThumbnails plugin with the same.

jneilliii, 2020-11-30:
Can you try with my RC version I had queued up to verify it fixes the issue please? Copy/Paste the URL below into plugin manager > get more > ...from URL and click Install.

```
https://github.com/jneilliii/OctoPrint-UltimakerFormatPackage/archive/0.2.1rc1.zip
```

jneilliii, 2020-11-30:
I just released these changes official.

closed 2020-11-30 by jneilliii

Jon-So, 2020-11-30:
it says rate limit exceeded for updating from the update server lol

Jon-So, 2020-11-30:
Yes it worked! thank you!, can you take over the webcam tab plugin and the touchui? lol I am joking, this new update broke allot of things why was it released so fast?

jneilliii, 2020-11-30:
A lot of people tested the release and there didn't seem to be any major issues reported during the release candidate cycle. I think TouchUI might already have fixes available. Not sure about the webcam tab plugin.

jneilliii, 2020-12-01:
BTW, you may like the new UI Customizer plugin, it has an option for putting the webcam in the right sidebar and can go full size.

jneilliii, 2020-12-01:
If you want to try I just patched together a fix for webcamtab that seems to work in 1.5.0 with my limited testing. You can install by copying/pasting the URL below into plugin manager > get more > ...from URL and clicking install.

```
https://github.com/jneilliii/OctoPrint-WebcamTab/archive/master.zip
```

Jon-So, 2020-12-01:
You are a God-sent!!!! why aren't you running the plugin library?????? 
thanks for the webcam tab code, I dont think the TouchUI creator is active he hasn't updated much lately... Someone else even figured out the problem to the autologin-loop problem, now I am having an issue with the icons not showing up... I guess time will tell...

jneilliii, 2020-12-01:
lol...I am one the plugin repo maintainers since I'm responsible for a bunch of plugins already. I think the issues with TouchUI are related to fontawesome upgrade in 1.5.0. He released a version 12 hours ago with 1.5.0 compatibility, you may just need to force an update check in software update to get the new version.

Jon-So, 2020-12-01:
Hey, I saw the update, thanks, unfortunately, he still hasn't fixed the login-loop, I had to manually go in and change the chromium.xinit file in order to make it work... Its a fix for the problem that allot of people are having, I am surprised he didn't add it to the update...

jneilliii, 2020-12-01:
Ah, that's a different beast outside of the plugin, I don't think they're tied together. Glad you got it straightened.

Jon-So, 2020-12-01:
Thank you, you are amazing...  about his plugin, its a file inside a side plugin he adds its called TouchUI- AutoStart. He maintains this as well...