jneilliii/OctoPrint-UltimakerFormatPackage #31: Feature Request: Display Image to the left of the list item

MaggiWuerze, 2020-12-01:
Could you add the option to place the image left of the list item instead of above?
that way we wouldn't have to increase the height of the list and stil have a thumbnail present.

jneilliii, 2020-12-01:
I tried that when initially making this plugin and it didn't work out so well, but I will see if I can do it now with 1.5.0 because the format changed a little and might be easier to accomplish now.

MaggiWuerze, 2020-12-01:
That's why I thought it could be done. There's a lot more horizontal space with 1.5.0
Thanks for the effort

jneilliii, 2021-02-15:
Not yet. My initial testing made me think it was going to take a little more logic, but I've recently injured myself so development time has been reduced a bit. I will be working on this eventually and will hopefully have an update in the near future.

MaggiWuerze, 2021-02-15:
Oh man I'm sorry. Hope you feel better soon. Maybe I'll have a look and make a PR for it.

jneilliii, 2021-03-07:
I've expanded the plugin with the above commit and released version 0.2.3rc1 for testing. You can change your release channel in OctoPrint's Software Update settings section to `Release Candidate` and upgrade when prompted,

MaggiWuerze, 2021-03-07:
Oh nice, I will definitely look into it and give you feedback

MaggiWuerze, 2021-03-08:
Ok, so Octoprint does not offer me an RC. Does it take time or do I need to do something else besides setting it to RC channel?

jneilliii, 2021-03-08:
What version of octoprint are you running?

jneilliii, 2021-03-08:
You can install manually by copying/pasting the url below into plugin manager > get more > ...from URL and clicking install. 

```
https://github.com/jneilliii/OctoPrint-UltimakerFormatPackage/archive/0.2.3rc1.zip
```

MaggiWuerze, 2021-03-08:
> What version of octoprint are you running?

OctoPrint 1.5.3
Python 2.7.16
OctoPi 0.17.0

MaggiWuerze, 2021-03-08:
> https://github.com/jneilliii/OctoPrint-UltimakerFormatPackage/archive/0.2.3rc1.zip

Will try once it finished printing

jneilliii, 2021-03-08:
I Just fixed the release, had tagged it to the wrong branch by mistake.

MaggiWuerze, 2021-03-08:
Cool, then I'll try that one in the morning

MaggiWuerze, 2021-03-09:
works like a charm
[image]

jneilliii, 2021-03-09:
Perfect, thank you. I will be releasing an official version in the near future, just want to get the other last issue resolved before release.

closed 2021-03-09 by jneilliii