jneilliii/OctoPrint-UltimakerFormatPackage #33: Cura not sending UFP file?

deputyd0ng, 2020-12-16:
I can send the print from octoprint to cura, but it is not sending the file with the UFP shown in the file name, and I have no new button for picture with the file.  I have in-line turned on.

Is there additional config required in cura?

jneilliii, 2020-12-16:
Are you using the Cura plugin `OctoPrint Connection` to send files to OctoPrint? It should auto-detect the plugin is installed and automatically upload in ufp format if it sees the plugin. Otherwise, when you create your file you have to change the type from gcode to ufp manually prior to upload.

deputyd0ng, 2020-12-16:
yes, I am sending the print directly from cura, but not getting the UFP tag, nor the extra buttons.

jneilliii, 2020-12-16:
What version of the OctoPrint Connection plugin do you have installed in Cura?

deputyd0ng, 2020-12-16:
3.5.17

jneilliii, 2020-12-16:
This is definitely strange. I just tested and it appears to be working fine here for me. I have the Cura plugin configured to connect to my OctoPrint instance (where this plugin is installed) and when I send the print the image is getting extracted.

[image]

[image]

Do you have any other plugins installed that interact with the file list to add additional buttons or anything there? This could be a conflict. Also, please upload your octoprint.log file so I can evaluate if there are any errors on extraction. You can just drag the file into a comment here to share. Of course, sanitize any personal information from that log you feel fit, ie passwords and such.

deputyd0ng, 2020-12-16:
so I saved it, and changed it to .ufp, and octoprint says its invalid.

deputyd0ng, 2020-12-16:
Seeing as how Ive gotten to at least this point, Im trying to use this to display the thumbnail of whats currently printing in Home Assistant, however, it seems the URL to pull the image isnt working, im using this url:

octoprint-ip-address:8080/plugin/UltimakerFormatPackage/thumbnail/

Would this be the correct URL, or has it changed?

jneilliii, 2020-12-16:
If the thumbnail never gets extracted then it would never be there to access, but that address should be 

```
octoprint-ip-address/plugin/UltimakerFormatPackage/thumbnail/<subfolder if any>/filename.png
```

If it's not getting extracted though then your image will never be there to load. To manually save a file in ufp format, in Cura in the save to file dialog you have to change the type.

[image]

deputyd0ng, 2020-12-16:
Yes, I manually saved it and verified cura is displaying it, Im not sure why the octoprint plugin isnt doing UFP, but ill work on it.

right now, Im having issues displaying the STL un HA, but thats not your software.

Thank you for your help

jneilliii, 2020-12-16:
Have you restarted Cura since adding the plugin to OctoPrint? It might not have detected the plugin installed if not.

closed 2020-12-16 by jneilliii

deputyd0ng, 2020-12-17:
yes, I can send to octoprint from cura, it doesn't go as a UFP when exporting right to octoprint, Im saving as UFP, and uploading and the thumbnails work, but the url is not working to feth the thumbnail in HA

jneilliii, 2020-12-17:
That's weird, not sure why the ufp format wouldn't be automatic for you. I'd have to check with @fieldOfView to see if anything changed. It seems to be working fine for me when I send to OctoPrint. 

That's too bad about the thumbnail not working in HA. I'd venture to say it might be related to access permissions maybe. That I'm unsure of. I do know that the relative url does get returned to the file listing API as that's how I'm getting the information from the web interface. Here's an example return of the file I uploaded earlier via `/api/files/local/<filename>`

```
{
    "date": 1608151170,
    "display": "CCR10_MagnetHook.gcode",
    "gcodeAnalysis": {
        "analysisFirstFilamentPrintTime": 12.915100179077854,
        "analysisLastFilamentPrintTime": 1373.8492599562314,
        "analysisPending": false,
        "analysisPrintTime": 1379.0683469327073,
        "compensatedPrintTime": 2228.669733931855,
        "dimensions": {
            "depth": 180.0,
            "height": 13.399999350309372,
            "width": 161.0970058441162
        },
        "estimatedPrintTime": 2228.669733931855,
        "filament": {
            "tool0": {
                "length": 1137.4754638671875,
                "volume": 2.735949116957283
            }
        },
        "firstFilament": 0.00051923515208768,
        "lastFilament": 0.9991729231419673,
        "printingArea": {
            "maxX": 171.19700622558594,
            "maxY": 200.0,
            "maxZ": 13.679999351501465,
            "minX": 10.100000381469727,
            "minY": 20.0,
            "minZ": 0.2800000011920929
        },
        "progress": [[0, 2228.669733931855], [0.00051923515208768, 1580.994999607242], [0.03182594392128237, 1511.2411712026576], [0.05342837230349493, 1441.6975317072004], [0.08866898098449892, 1372.2081089589249], [0.11054159661506244, 1302.7665016147614], [0.14753412848097294, 1233.3033400429697], [0.18472021747100772, 1163.7362008177645], [0.22290447988954692, 1094.1700116298414], [0.2603885015177389, 1024.7246496004243], [0.29858201239954685, 955.2750979707237], [0.33519998150307345, 885.8347675921079], [0.38403847360719795, 816.3979015155538], [0.4299082090020578, 746.9590010344015], [0.48241503800127494, 677.5189433470753], [0.5396676498663927, 608.0062261101983], [0.6012459001232028, 538.5658490465706], [0.6620941823863017, 469.1140173498253], [0.7163575588020598, 399.65603716415893], [0.7573355177322768, 330.21813464370257], [0.8160328716808751, 260.7665817793603], [0.8725819397330498, 191.32576624893608], [0.9254785253985922, 121.59375336229235], [0.9742251934745486, 52.13637545539282], [0.9991729231419673, 6.022313796437811], [1, 0]]
    },
    "hash": "0ebea32c0eff6a37ad1e9e5ee3f132921b05dd9c",
    "name": "CCR10_MagnetHook.gcode",
    "origin": "local",
    "path": "CCR10_MagnetHook.gcode",
    "refs": {
        "download": "https://vlx.kicks-ass.net/downloads/files/local/CCR10_MagnetHook.gcode",
        "resource": "https://vlx.kicks-ass.net/api/files/local/CCR10_MagnetHook.gcode"
    },
    "size": 1513765,
    "thumbnail": "plugin/UltimakerFormatPackage/thumbnail/CCR10_MagnetHook.png?20201216153930",
    "thumbnail_src": "UltimakerFormatPackage",
    "type": "machinecode",
    "typePath": ["machinecode", "gcode"]
}
```

The last little bit at the end of the png filename is just a timestamp cache buster in case you upload the same file so it will update the image in the UI.

fieldOfView, 2020-12-17:
@deputyd0ng do you have the "Store Gcode on printer SD card" option selected? If so, the plugin does not upload in the UFP container. @jneilliii, would storing on the SD card be compatible with the UFP upload?

deputyd0ng, 2020-12-17:
> @deputyd0ng do you have the "Store Gcode on printer SD card" option selected? If so, the plugin does not upload in the UFP container. @jneilliii, would storing on the SD card be compatible with the UFP upload?

I dont have that option checked(although, for resume from power-loss, it may be useful), Ive literally never printed from SD card Ive only ever used mainsail or octoprint.  TBH Im less concerned about sending it directly to octoprint, I dont mind saving as UFP and uploading, I just want the STL preview to appear in my HA page.

@jneilliii should I be able to access that url from my browser to verify it works, and then plug it into HA?

deputyd0ng, 2020-12-17:
> That's weird, not sure why the ufp format wouldn't be automatic for you. I'd have to check with @fieldOfView to see if anything changed. It seems to be working fine for me when I send to OctoPrint. [...]

I think youre right, if I access the thumbnial URL in my browser, I get a "403 forbidden"

jneilliii, 2020-12-17:
> would storing on the SD card be compatible with the UFP upload?

no it most certainly would not, pretty sure I'm only processing files with a destination of "local"

> I think youre right, if I access the thumbnial URL in my browser, I get a "403 forbidden"

I was thinking this might be the case. You may be able to get around it by using the AutoLoginAs settings/plugin to enable automatic login to OctoPrint from the local network. The plugin can be found here, and is just an easy way to set those values from the web interface. https://plugins.octoprint.org/plugins/autologin_config/

fieldOfView, 2020-12-17:
How about if you try `/api/files/local/<filename>?apikey=<your api key>`

fieldOfView, 2020-12-17:
What distribution of Cura do you use? On what OS? If you are on a Linux, try using the official AppImage instead of the distribution in your OS package manager.  The latter may not come with all required dependencies.

jneilliii, 2020-12-17:
> How about if you try /api/files/local/<filename>?apikey=<your api key>

good idea. i didn't realize you could append apikey to the get request as a query parameter, figured it had to be in the header.

BenBergman, 2021-01-30:
> What distribution of Cura do you use? On what OS? If you are on a Linux, try using the official AppImage instead of the distribution in your OS package manager. The latter may not come with all required dependencies.

Just noting that switching from the Arch Linux `cura` package to the `cura-appimage-bin` package from the AUR fixed this issue for me. Cura is now sending UFP files to OctoPrint instead of raw gcode.

meteozond, 2021-02-18:
I've spent some time figuring out similar problem and found that my Cura `UFP Writer` was disabled for unknown reason. Turning on and Cura restart fixed it.

[image]

BenBergman, 2021-02-18:
I'm assuming @meteozond 's suggestion is for non-Linux folks, but just in case I  switched back to the Arch Linux `cura` package and checked for the UFP Writer plugin. It was already enabled and UFP files are still not being written.

fieldOfView, 2021-02-18:
As far as I know, the Arch linux package is broken. It does not have a required dependency (libcharon) which is needed for UFP files to work. Use the AppImage instead (or get the Arch linux package fixed).

BenBergman, 2021-02-18:
I've filed a bug report is the Arch issue tracker (https://bugs.archlinux.org/task/69692). I am unable to work on fixing it myself right now, but hopefully that at least moves the ball forward a bit.

TTSKarlsson, 2023-04-05:
I know this is closed, but I just solved one reason for this problem that could be added to some FAQ or future support maybe.

I had PrintJob Naming by FieldOfView installed, after removing that I got thumbnails enabled files sent to Octoprint.

jneilliii, 2023-04-05:
I'd recommend using my other slicer thumbnails plugin now a days. I'm no longer really working on this one and with the thumbnail post processing script now available in cura by default there's no reason not to use it.

TTSKarlsson, 2023-04-05:
Thanks, I had no idea =)