jneilliii/OctoPrint-UltimakerFormatPackage #35: Thumbnails without Cura Plugin?

omgsh, 2021-02-01:
I am running more than one printer so every time I want to send a file to one of them, I have to swap the Octoprint instance Cura is using as I believe printer profiles do not store a selected Octoprint instance, Cura only defaults to the last used instance no matter if that instance was used for a different printer profile or not.

Saving to file and uploading through Octoprint or Octofarm is a logical fix as it both saves me from the extra steps of changing Curas Octoprint instance and allows me to catalog the file in a folder system locally before uploading.

I believe there is a way to save files with the thumbnail data by selecting the Ultimaker 2 Gcode option in Cura Preferences (although they have not displayed thumbnails when I tried) this is also an issue as my printers (Ender 3 V2) use Marlin. 

Is there a way to save the file with Cura to include the needed data that my printers can use or is there a step I might be missing with any of this?

omgsh, 2021-02-01:
Cura is not remembering the instance I set for each printer so I think this can be solved.

closed 2021-02-01 by omgsh

jneilliii, 2021-02-01:
I see you closed this, so I assume you figured out that the octoprint connection settings are tied to your Cura printer profiles, so you should be able to have both. The other option would be to save your sliced file locally in ufp format and upload that file to OctoPrint.

fieldOfView, 2021-03-29:
> Cura is not remembering the instance I set for each printer so I think this can be solved.

Please consider reporting this at the OctoPrint Connection plugin issue tracker instead of here:
https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues

fieldOfView, 2021-03-29:
To answer your question: Use File -> Export and export as Ultimaker Format Package (*.ufp).