jneilliii/OctoPrint-UltimakerFormatPackage #39: Gcode Viewer and progress bar displaying incorrectly

jimmyeao, 2021-03-02:
Strange issue after installing:
[image]

Gcode viewer skips to 100%, 
Progress shows over 100% 
Layers show correctly

other plugins I have:
[image]

jneilliii, 2021-03-02:
Does it do the same with Themeify disabled?

jimmyeao, 2021-03-02:
I’ll test in the morning, printers busy at the moment :)

jneilliii, 2021-03-02:
I just tested on my install without Themeify and everything appears to be working fine. The bottom bar progresses during layer print progress and the right bar goes up with the print. I'm not saying it's Themeify is the issue, but am currently not able to reproduce.

[image]

[image]

jimmyeao, 2021-03-03:
Disabling themify seemed to do it, so I'll go through the themify settings one by one and see if I can narrow it down