jneilliii/OctoPrint-UltimakerFormatPackage #41: no previews in custom folders

Goodfeat, 2021-03-12:
Hello, I noticed a problem
if you create a new folder in octoprint and put a file in it, then a preview of the model does not appear inside this folder

[image]

OctoPrint 1.4.2 Python 2.7.16 OctoPi 0.17.0

jneilliii, 2021-03-12:
I can't say I'm seeing the same thing on my end. I just tested by uploading a file to a sub folder and the thumbnail loaded up just fine. Please confirm you're uploading a ufp formatted file.

[image]

Goodfeat, 2021-03-14:
Yes it works but

you need to manually save the file with the .ufp extension so that you can save it in a separate folder. If you just copy the g-code, then it does not work(

I think I understand what is the peculiarity of this script.
as I understand it, the .ufp format splits the file into two parts
one part is a preview
the second part is directly the G-code

In this case, is it possible to somehow save the preview directly in the G-code?
I'm sure there is a way to somehow convert the image and save it in G-code using the Octoprint script
после этого выводить изображение из файла G-codethen output image from G-code file

a similar method was implemented in Prusa slicer
[image]

jneilliii, 2021-03-14:
There is a post-processing script approach that will allow you to use Cura and the PrusaSlicer Thumbnails plugin. Follow the instructions on the comments at the link below.

https://gist.github.com/jneilliii/4034c84d1ec219c68c8877d0e794ec4e

jneilliii, 2021-03-14:
The other option is sending the file directly from Cura using the OctoPrint Connection plugin from Cura's Marketplace. It will detect the presence of this plugin and automatically send files to OctoPrint in ufp format.

Goodfeat, 2021-03-14:
Thanks you,
i will try this way

Goodfeat, 2021-03-14:
unfortunately I was not able to use the script.
there is probably an error in the script and it does not appear in the list of plugins

checked on chicken 4.5, 4.8, Cura Arachne_engine_alpha 99.9.0

[image]
[image]

the same on the other version of the Cura
the instruction is simple, it's hard to make a mistake, but it really doesn't work

Goodfeat, 2021-03-14:
Found an error, the error was in the class name

jneilliii, 2021-03-14:
Thanks, I've updated the file name in the gist.

closed 2021-04-05 by github-actions[bot]