jneilliii/OctoPrint-UltimakerFormatPackage #42: Image lost while moving uploaded files in other folders

hawkeyexp, 2021-05-14:
If you reorganize your uploaded files inside octoprint via the gui the image path seems to be not updated and the preview image is lost.

I have seen there is a metadata.json for each uploaded file. Is there any chance to implement a routine etc. which checks the uploaded files and metadata.json and possible fix it after file moves ?

jneilliii, 2021-05-14:
If it's not also updating the metadata.json file automatically then you have come across a new bug. The FileAdded/FileRemoved events in the code should be handling this properly unless something changed. Can you give more details as to the process you used to move files and what version of OctoPrint you are running?

hawkeyexp, 2021-05-15:
I trid to reorganize my folder scructure via ne octoprint gui. I created a new folder there, selected the old folder which i want to move via the scissors symbol to move it - you get the list of folders and i moved my old folder including the old files as subfolder to the new one.
I think i found the problem with your hint and reproducing during this typing :-)

It seems to work for moving single files but sometimes fails while moving complete folders. During typing i checked it again and first all was working but now all images fails again - i was moving my test folder containing ~ 15 files from subfolder (3. level) to root level and boom :-)

Ocotprint is 1.6.1 but it happend to me also with 1.5.3 which i was using before - i didn't move again since my first fail which crashed all previews by tried again tonight and it was no good idea :-)

jneilliii, 2021-05-15:
Thanks, you might be right, folders could be problematic as I'm not sure how OctoPrint's events work in that instance. If you look in the plugin's data folder (typically `~/.octoprint/data/UltimakerFormatPackage`) do you see the same folder structure as the new or the old, and are there any errant thumbnails in the old path that didn't get moved to the new path or are they all just gone?

hawkeyexp, 2021-05-15:
if i check the data folder i only can see the old folder structure for old files. If i upload new files the new structure is created.

jneilliii, 2021-05-15:
perfect, thanks. that's kind of what I expected. but curious are the png files still there or were they deleted in the old structure?

hawkeyexp, 2021-05-15:
some was left but most was deleted

jneilliii, 2021-05-15:
Thanks for doing all the leg work and all info. I'll be looking into it. This is one of the reasons the PrusaSlicer Thumbnails plugin works better overall since the image is embedded in the gcode file itself. There's also a post-processing script bundled with Cura 4.9 that will enable the right format for the other thumbnail plugin. I'll be renaming that one soon...

jneilliii, 2021-05-15:
I've got the initial patch work done in the above commit. Still needs a little work in regard to updating file metadata. It will move the subfolder with thumbnails based on my initial testing. I also updated the scan files button in the settings that will allow for the moved images to be seen and update the metadata that way.

hawkeyexp, 2021-05-15:
wow - you are really fast! Let me know when it's ready to test.

jneilliii, 2021-05-15:
Logic was basically the same as the file move logic, just had to do a little other tweaks. I've gotten 1.0.0rc1 released now if you want to test it out. I've been moving a single folder with only files, not sub-folders so far, about to test subs of subs, but if you want to give it a go it's there for you.

jneilliii, 2021-05-15:
Initial testing of moving complex folder structure seems positive. Let me know what you see on your end.

hawkeyexp, 2021-05-15:
I will give it a try tomorrow - an multiple hour print should run over night :-)

closed 2021-05-19 by jneilliii