jneilliii/OctoPrint-UltimakerFormatPackage #45: Preview does not update when changing files

ldursw, 2021-06-01:
There is a regression between 0.2.2 and 1.0.0.

When I select a file the preview image is shown in the State panel. If I select another file the information is updated accordingly but the preview doesn't update. Refreshing the page shows the correct image.

Version 0.2.2
https://user-images.githubusercontent.com/37294448/120399125-c6acb600-c311-11eb-9fd3-7cfc7badba29.mp4

Version 1.0.0
https://user-images.githubusercontent.com/37294448/120400693-c530bd00-c314-11eb-9724-4f16539bf474.mp4

dumbozo, 2021-06-02:
same issue here.
Has worked in the past.

jneilliii, 2021-06-02:
Thanks for the report. This does appear to be a regression of some kind. I'll look into fixing this soon.

jneilliii, 2021-06-02:
Fixed by the above commit and available in version 1.0.1rc2. If you would like to test, please change the plugin's release channel in OctoPrint's Software Update settings to `Release Candidate` and install the update when prompted.

ldursw, 2021-06-02:
Thanks. It is working now.

PR #46 (2021-06-02): 1.0.1
* fix issues related to odd file naming, resolves #34
* fix incorrect jQuery selector for image when switching files, resolves #45