jneilliii/OctoPrint-UltimakerFormatPackage #48: Image Button Missing

mickmolloy, 2021-08-15:
Pi 3b+
Cura Thumbnails Version: 1.0.1
OctoPrint: 1.6.1 
Cura 4.10.0
OctoPrint Connection 3.5.18

I am not getting the newly added image button in octoprint following the install of the plugin , have restarted both Cura and the Pi but no change. Fresh install of the Octopi tonight

Thank You

mickmolloy, 2021-08-15:
Note the Thumb works but I am not getting the newly added image button in octoprint,

mickmolloy, 2021-08-15:
[image]

jneilliii, 2021-08-15:
the button only shows up if you are not showing the image in the file list. In that case, if you click the image it should open the pop-up.

mickmolloy, 2021-08-15:
so it will always show the thumb now?

jneilliii, 2021-08-15:
if that option is enabled for it to show in the file list, yes. the option is in settings for you to adjust to your liking.

jneilliii, 2021-08-15:
it only works on new files that you have uploaded since installing the plugin.

jneilliii, 2021-08-15:
for example, this is how I have my Slicer Thumbnails plugin currently configured and the options are the same.

[image]

mickmolloy, 2021-08-16:
Thank you for the help and quick replies, the problem is solved. I just need to send the file from cure to octoprint and it works.

Cheers

Mick

closed 2021-08-16 by mickmolloy

jneilliii, 2021-08-16:
Yeah, this plugin does require the format to be ufp not gcode. The cura collection plugin does that automatically for you when it sees my plugin installed.