jneilliii/OctoPrint-UltimakerFormatPackage #54: Feature Request - State panel thumbnail and text align options

spaded21, 2022-01-29:
Would it be possible to add align options to the thumbnail in the state panel? Or some other way to force the text to stay underneath the thumbnail? I find that I need to scale the thumbnail to 89% or higher otherwise the text tries to squeeze in on the side, making it difficult to read. I would prefer to keep the thumbnail somewhere around 60% so that I can see more of the File List Panel underneath it. This is especially a problem when the filename uses underscores instead of spaces because it won't break up the lines and wastes even more space.

Thank you for an otherwise excellent plugin.

[image]

jneilliii, 2022-01-29:
I'm sure there's some kind of css trick I could use to handle this, will just take some experimentation. Looks like you have UI Customizer of Themeify installed, could potentially use their advanced/custom css options to in the interim.

jneilliii, 2022-01-30:
Added with above commit and part of version 1.0.2rc2. Change the plugin's release channel in OctoPrint's Software Update settings to `Release Candidate` and update when prompted. 

[image]

spaded21, 2022-01-30:
Thanks so much for such a quick response. I will install it once my current prints are done.

spaded21, 2022-01-30:
Worked perfectly, I just had to load a new file to see the change. Thanks again!

PR #50 (2022-04-16): 1.0.2
* fix special characters issue in OctoPrint 1.7.0, resolves #49.
* add break after image option for state panel, resolves #54
commit 5bc5d33c (2022-04-16): 1.0.2 (#50)

* fix special characters issue in OctoPrint 1.7.0, resolves #49
* add break after image option for state panel, resolves #54
closed 2022-01-30 by spaded21