jneilliii/OctoPrint-UltimakerFormatPackage #55: Not Working Over Reverse Proxy

BzowK, 2022-02-24:
Hey Guys - 

Love this plugin, but it doesn't work for me when accessing OctoPrint over reverse proxy.  Everything else in OctoPrint works over RP. Was hoping someone could advise.  

Below are the URLs of the images (or lack thereof) copied from Chrome when accessing via IP vs RP.  Visiting the top directly yields an image, but the bottom results in a 400 - Bad Request.
- http://192.168.0.85/plugin/UltimakerFormatPackage/thumbnail/Glasses%20Holder%20(Test2)--0.4_0.2mm_15.0%_190.0C_65.0mms_2g_0-21.png?20220219103322
- https://mydomain.com/octoprint/plugin/UltimakerFormatPackage/thumbnail/Glasses%20Holder%20(Test2)--0.4_0.2mm_15.0%_190.0C_65.0mms_2g_0-21.png?20220219103322

I use nginx and below is the config I use...
		location /octoprint/ {
			proxy_pass http://192.168.0.85/;
			proxy_set_header Host $http_host;
			proxy_set_header Upgrade $http_upgrade;
			proxy_set_header Connection "upgrade";
			proxy_set_header X-Real-IP $remote_addr;
			proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
			proxy_set_header X-Scheme $scheme;
			proxy_set_header X-Script-Name /octoprint;
			proxy_http_version 1.1;
			client_max_body_size 0;  
        }

Any suggestions?  Thanks!

jneilliii, 2022-02-24:
Paths definitely look like they are correct for your reverse proxy implementation. I use the plugin on default octopi install which uses haproxy as its reverse proxy without issue. Maybe there's extra bits you have to add to nginx? I wouldn't think so. Any errors in octoprint log?

BzowK, 2022-02-24:
Thanks for the reply...

Yeah, I was thinking that might be the case which is why I posted my config.  I looked up a few OP config examples for nginx, but didn't see anything I didn't have which stood out.  The nginx log didn't contain any errors, either; which I forgot to mention initially.

Any suggestions would be appreciated - Thanks!

closed 2022-03-18 by github-actions[bot]