jneilliii/OctoPrint-UltimakerFormatPackage #56: Question: Can I change the color of the model in thumbnail?

Kirbo, 2022-03-24:
As it reads in the title:
Can I change the color of the model in thumbnail?

Currently the models are always yellow, but if I wanted to change the color for example violet: can I do this somehow? And where should it be possible to change? In Ultimarek Cura settings somewhere or this Plugin, or...?

Thanks for the awesome plugin either way! 🙂

jneilliii, 2022-03-24:
I'm not entirely sure it's possible in cura. I believe in PrusaSlicer the color of the image is controlled by the filament color configuration.

fieldOfView, 2022-03-24:
In Cura the material defines the color of the model in the thumbnail (and in the viewport)

jneilliii, 2022-03-24:
Thanks @fieldOfView

closed 2022-03-24 by jneilliii

JaneX8, 2023-11-27:
For reference: https://3dprinting.stackexchange.com/questions/21567/why-are-my-cura-generated-thumbnails-showing-in-one-flat-color-without-any-depth.