jneilliii/OctoPrint-UltimakerFormatPackage #57: Error during up grade from 1.0.1 to 1.0.2

StarNamer, 2022-04-18:
From octoprint.log:

Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/plugins/softwareupdate/__init__.py", line 2089, in _perform_update
    update_result = updater.perform_update(
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/plugins/softwareupdate/updaters/pip.py", line 115, in perform_update
    returncode, stdout, stderr = pip_caller.execute(*pip_args, **pip_kwargs)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/util/pip.py", line 208, in execute
    return self.call(command, **kwargs)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/util/commandline.py", line 244, in call
    all_stdout += process_stdout(p.stdout.readlines(timeout=0.5))
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/util/commandline.py", line 236, in process_stdout
    return process_lines(lines, self._log_stdout)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/util/commandline.py", line 229, in process_lines
    processed = self._preprocess_lines(
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/util/pip.py", line 498, in _preprocess_lines
    return list(map(self._preprocess, lines))
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/util/pip.py", line 517, in _preprocess
    return to_unicode(clean_ansi(text))
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/util/commandline.py", line 57, in clean_ansi
    return _ANSI_REGEX.sub(b"", line.encode("latin1")).decode("latin1")
UnicodeEncodeError: 'latin-1' codec can't encode characters in position 5-43: ordinal not in range(256)
2022-04-18 10:51:10,279 - octoprint.plugins.softwareupdate - INFO - Saved update log to disk
2022-04-18 10:51:11,210 - octoprint.plugins.tracking - INFO - Sent tracking event update_failed, payload: {'target': 'UltimakerFormatPackage', 'from_version': '1.0.1', 'to_version': '1.0.2'}

jneilliii, 2022-04-18:
I think this is actually a pip error and the plugin probably upgraded fine. IIRC this issue is fixed with 1.8.0 octoprint which will be released in the near future and currently in rc phase. If you restart octoprint what version stored for the plugin?

StarNamer, 2022-04-18:
You are right. After a restart Octoprint shows version 1.0.2 installed. Thanks.

closed 2022-04-18 by StarNamer