jneilliii/OctoPrint-UltimakerFormatPackage #58: "State" thumbnail only showing when you're in the same folder as the file to print

JMVS, 2022-05-01:
Hello.

It seems I've found a problem when switching folder and having a file selected for print in which the thumnail disappears when you're not in the same folder as the file you've selected. I've tested this with two instances of OctoPrint & Cura Thumbnails and the results are the same.

OctoPrint: 1.7.3
Cura Thumnails: 1.0.2

**When you're in the same folder as the file selected for printing:**
[image]
[image]

**When you have the same file selected but move to a different folder:**
[image]
[image]

_Cura Thumbnails Settings:_
[image]
[image]

Let me know if you want me to run any other tests.

jneilliii, 2022-05-02:
Huh, I thought I had fixed this issue, but it may have been in my Slicer Thumbnails plugin instead. You could potentially try that one, as I've been considering retiring this one since that one supports most major slicers.

fieldOfView, 2022-05-02:
But that does not integrate as automatically with the OctoPrint Connection plugin as this plugin does.

jneilliii, 2022-05-02:
That's true, you have to enable the post-processing script manually, but everything after that works with the OctoPrint Connection plugin without issue.

fieldOfView, 2022-05-03:
I'll see if I can directly support the other plugin as well (ie: include the thumbnail with the gcode if that plugin is detected)