jneilliii/OctoPrint-UltimakerFormatPackage #60: Colour of thumbnails

TheCableGuy99, 2022-09-24:
Hi,

I'm just having an issue where all my thumbnails are black and have no colour to them.

Is this a setting in Cura or the plugin i'm missing?

Thanks.

TheCableGuy99, 2022-09-24:
I found that it's creating the thumbnail in the colour of the filament selected, I didn't spot it before because I tested that by changing the colour and it still came out black... I didn't realise you had to restart Cura for it to take effect.

Anyway, would it be possible to add an option to either override certain colour hex codes with another colour? (Because I use black a lot and it doesn't look that nice in the plugin). And/or add an option to override ALL colours with another colour?

Thanks for your consideration.

jneilliii, 2022-09-24:
I'm not going to get into image manipulation with the plugin.

closed 2022-09-24 by jneilliii