jneilliii/OctoPrint-UltimakerFormatPackage #62: Black/Shadowed thumbnails

Peabuttt, 2022-11-20:
I'm guessing the issue is probably between the chair and the computer, but here it goes... all my thumbnails are just like a shadow of the 3d model. check attachment. How can I fix it?
[image]

jneilliii, 2022-11-20:
The color of the thumbnail is based on the color of your filament profile. I can't control that unfortunately.

closed 2022-11-20 by jneilliii

Peabuttt, 2022-11-20:
So i googled and i also found that. I changed the colo
[image]
r of the filament in cura just to test it (yellow). The result was the same. Maybe i changed something i thought it was the filament but it wasn't.