jneilliii/OctoPrint-UltimakerFormatPackage #65: State thumbnail only showing when it is not hidden

dekkerso, 2024-01-15:
When hiding the file which is printing, the thumbnail is not showing. This might be related to the bug where the thumbnail is not showing when in a different folder. Only showing in "state" when it is visible in "files"?

[image]

[image]

[image]

jneilliii, 2024-01-15:
So you start the print that was previously successful and then filter the file list to not show successful prints?

dekkerso, 2024-01-15:
that filter is always on. when i reprint that print it will not show the thumbnail.

closed 2024-02-06 by github-actions[bot]