jneilliii/OctoPrint-UltimakerFormatPackage #66: dumb question

xmodpt, 2024-01-18:
is it possible do adapt the plubin but for resin printers ?

For files from Chitubox like CTB (elegoo) photon (.photon) etc...

Kind regards
Nuno

jneilliii, 2024-01-18:
No idea how that would work but would be more apt to do it in my slicer thumbnails plugin that is more generic and works with multiple slicers. Not really developing this plugin anymore because of that.

closed 2024-01-18 by jneilliii

jneilliii, 2024-01-18:
I'm curious what @rudetrooper thinks about the possibility of this. He's more familiar with the format since he maintains the chituboard plugin.

xmodpt, 2024-01-18:
i know i am the one how did the videos that he has in the git however last time o spoke with him... he had no to little time to work on this an i am trying to keeps this alive.... for exemple the plugin does not work with octoprint 1.9.x and i found a way to revert to 1.8.7 and keep the plugin working. i have posted the solution in his github. i am will at some point in time release a new set of videos for this..

thank you np